Ultimaker/Cura #8416: Warning for severe speed reduction caused by "minimum layer time"

myfreeweb, 2020-09-20:
**Is your feature request related to a problem? Please describe.**
When e.g. printing a [calibration cube](https://www.thingiverse.com/thing:1278865) at very high speeds with low infill, the behavior can extremely surprising: the printer *slows down* after the bottom layers. Even though the easily discoverable settings talk about speeding up after bottom layers. I first suspected firmware (!) but checked the gcode and indeed the bottom layers had much higher feed rates. I had to find [this stackexchange comment](https://3dprinting.stackexchange.com/questions/13754/how-to-make-cura-generate-g-code-with-desired-feed-rate#comment24630_13754) to realize that "minimum layer time" is a thing and I think it was enabled by default. It was *hidden* by default too.

**Describe the solution you'd like**
Display a warning when minimum layer time is affecting the emitted feed rates.
Maybe highlight which layers are slowed down.
Just somehow make that setting much more obvious.

**Affected users and/or printers**
Anyone who wants to test high speeds on their printer.

enly1, 2020-09-22:
Just an extra snippet which can help when you know it's a thing - you can set colour scheme to speed.  That allows you to see the speed changes per layer quite easily.

Liger0, 2020-09-22:
it's harder than that, since the color also changes on the speed scheme because of acceleration, so it's hard to tell if the limitations are because of the acceleration not being enough or because of the minimum layer time. It can a bit hard for materials like abs, where slowing too much ends up melting everything.

Ghostkeeper, 2020-09-23:
Minimum layer time kicks in on most prints. Only prints with a flat top side are exempt, since if the top side is not flat the last few layers will pretty much always hit the minimum layer time. I don't think displaying a warning for something that's expected to improve quality is appropriate, and displaying that warning with almost every slice just becomes annoying.
There are many things that Cura does to improve print quality or productivity, without necessarily notifying the user of it: limiting the number of retractions, printing at various line widths, removing tiny pieces, or even basic things like retracting the filament and printing support. It takes a lot of effort to learn these things, but according to the profile you're using the end result is better off, so you'll not normally need to know that this happens at all.

> it's harder than that, since the color also changes on the speed scheme because of acceleration, so it's hard to tell if the limitations are because of the acceleration not being enough or because of the minimum layer time.

No, the colours are not taking acceleration into account, just cruise speed. It would be nice if it did show accelerations, but the issue is still open: https://github.com/Ultimaker/Cura/issues/7401