Ultimaker/Cura #8418: PETG: extensions in Cura / Experimental / Wipe nozzle

stonefred, 2020-09-21:
Hello developers,

I use the 3D printer "anycubic-predator":
https://www.anycubic.com/products/anycubic-predator-fdm-printer

The filament here is PETG. That doesn't warping. But after a while the nozzle sticks (the nozzle was new and clean...). 

[image]

Which unfortunately slowly but surely destroys the part!

[image]

A cleaning brush cannot be attached to the X-axis (shown in red in the picture) of the delta printer. 

**A) Further two optional input values are required:**
1. additionally required: **Z absolute** (shown in green in the picture)
2. additionally required: **Y absolute** (shown in blue in the picture)
This gives an absolute starting position X-Y-Z for the wipe move (shown in orange in the picture).

Description of a cleaning procedure for Delta printers:
1. Movement A (magenta)
2. Movement B (red)
3. Multiple wipe movements (yellow)
4. Movement B backwards (red)
5. Movement A backwards (magenta)

The cleaning brush is mounted on the frame of the delta printer and therefore has an absolute position X Y Z (orange).

**Avoiding collisions between the nozzle and the printed part:** 

If the brush is positioned lower than the top of the part, **a collision could occur**:

[image]

If the brush is mounted over the part there is no collision possible:

[image]

With every cleaning procedure, Cura should therefore check whether the currently printed layer is lower than the height (Z absolute) of the brush. If not, the cleaning procedure should no longer be carried out!

[image]

**B) Two additional optional cleaning procedures would be useful:**
1. **After "Start G-code"**   (No more waiting for the print to manually clean the nozzle!)
2. **Before "End G-code"**  (the nozzle is clean even after printing)

I think these enhancements are important for users of delta printers.
Some Cartesian 3D printers also require an additional axis. 

[Gehaeuse Temp-Steuerung  EMF.zip](https://github.com/Ultimaker/Cura/files/5245123/Gehaeuse.Temp-Steuerung.EMF.zip)

I hope this extension can be implemented in Wipe.

gr5, 2020-09-21:
Improvement to this feature idea: Instead of never-ever going *down* to the brush - you could go down by the "gantry height" as set in machine settings.  This would then also work nicely on say an Ultimaker with a brush mounted on the build plate in the corner at say 70mm above the glass.  You can still go "down" do the brush as long as the part won't hit the gantry.  For a delta printer the gantry height would be quite small - maybe 10mm.  Combine this with the head dimensions as you don't want that to hit the part being printed either.

gr5, 2020-09-21:
This feature could be written as a simple post processing plugin that had values like location of brush in XYZ and how much to allow the brush to be below the printed layer (defaults to zero).  And it could have a value in minutes of how often do wipe as the timing is in the comments of gcode files (each layer time is stored at the start/end of each layer).

Ghostkeeper, 2020-09-23:
This is similar to https://github.com/Ultimaker/Cura/issues/7235, but expands upon the idea (so I'll direct discussion from there to here).

In general I'm not happy with how little configuration there is for the wipe. It doesn't cover the wide range of 3D printers out there by a long shot.

SemperDiscere, 2022-07-30:
Any movement on this?