Ultimaker/Cura #8428: Looooong at launch !

cWal00, 2020-09-22:
15 minutes to get Cura to display is quite not normal !
Never had this problem before. Just installed last 4.7.1 release on Windows 10 Pro 2004 64. Once displayed it seems to work normally. But 15 minutes to open a program installed on a NVMe SSD is weird...
I tried uninstall/reinstall, I tried to delete the plugins folder, reboot laptop (Lenovo Legion Y540)... Same issue.

fieldOfView, 2020-09-23:
That is certainly way too long; about 60 times as long as it should be. Could you please share your Cura.log, so we can see where in the process Cura spends so much time?

One thing that caused unexpected delays in the past is huge networks. Does the same thing happen if you are not connected to the network?

Please next time fill in the issue template.

cWal00, 2020-09-23:
[cura yesterday.log](https://github.com/Ultimaker/Cura/files/5271871/cura.yesterday.log)
[cura.log](https://github.com/Ultimaker/Cura/files/5271872/cura.log)

Thanks for answering. You will find 2 log files attached. The 2nd is from today, since yesterday I uninstalled all previous versions that were on the hard disk, just in case, but the issue is still present, about 15 minutes to open.
I hope it will help.

nallath, 2020-09-24:
Woah. For some reason the loading of plugins takes a _very_ long time. ` Loading all plugins took 511.69245886802673 seconds`

But all other things are also taking ages. The only thing I can think of is that your hardrive is extremely slow.

cWal00, 2020-09-24:
It could have been an explanation, but unfortunately no, my hard drive is not slow : SSD NVMe 1TB, and the laptop has 32GB RAM. For example, Maya 2020 takes about 20s to launch, and it is one of the biggest program I use...
I send you the log of another install of Cura 4.7.1 on an older laptop (16GB) with a slower hard drive (SSD M2). Yet this one takes only around 30s to open...
[cura.log](https://github.com/Ultimaker/Cura/files/5275100/cura.log)

fieldOfView, 2020-09-24:
Is your 1st laptop connected to a network? Does it have network drives mapped? Do you have a drive labeled X: ? It seems like Cura accessing files is taking extremely long, like it is trying to reach a network drive for each file it is trying to open.

cWal00, 2020-09-24:
Yes, but they are both connected (though not to the same network, I'll try to connect to the connect to the same to check), they have both several drives mapped, and Cura is installed on the same partition. Here is a sreenie of window explorer of the 2nd laptop, working fine :
[image]
The X: drive is the drive embedded in my modem-router, not connected on this laptop. I'll try to remove it from the 1st laptop at home and see...

fieldOfView, 2020-09-24:
Using a different driveletter for the X: drive will probably make a huge difference.

cWal00, 2020-09-24:
Well, you were right ! Disconnecting the X mapped drive solved the problem, and even by reconnecting it using a different letter, launch is still OK : 10 sec to open !
Here is the cura log file FYI (correct start is at the end).
[cura.log](https://github.com/Ultimaker/Cura/files/5278629/cura.log)

Thanks again for your help, I am happy it is solved and that it didn't come from the software :-)

fieldOfView, 2020-09-24:
>  I am happy it is solved and that it didn't come from the software :-)

Well, it kind of sort of is the software though, but I am glad the workaround of not using X: as a driveletter for a network drive works. 

Could you close the issue?

closed 2020-09-24 by cWal00