Ultimaker/Cura #8430: [4.6.2-4.7.1] Visibility Settings Ticks not saved + Release Note/Changelog page always opening

WPT-git, 2020-09-23:
**Application version**
4.6.2; 4.7.1

**Platform**
Host: wpt-mint Kernel: 4.15.0-70-generic x86_64 bits: 64 compiler: gcc v: 7.4.0
Desktop: MATE 1.22.0 wm: marco dm: LightDM Distro: Linux Mint 19.2 Tina
base: Ubuntu 18.04 bionic 

**Printer**
does not matter

**Reproduction steps**
1. Downloading new cura version as appimage
2. appimage-file properties "run file as program" (gui version of chmod +x; no difference how it is done)
3. open Cura version
4. Release Note / Changelog screen appears
5. go to configuration/settings to tick additional visibility settings e.g. draft shield (and its height)
6. activate draft shield in the print/slicing settings and type 50 as value into the height field
6. close Cura
7. re-open Cura

**Screenshot(s)**
n/a

**Actual results**
- Release Note / Changelog popup appears on every reopening of Cura
- marked/ticked visibility settings are not visible on a new cura restart at the printing/slicing settings
- the on last run entered values of the fields, which were set to visible before, are saved and still there after cura restart; but ofc
only visible after the field was set to visible again.

**Expected results**
- Release Note popup not reappearing after the program had been opened once already and the popup was confirmed
- visibility settings staying visible and will not get unticked again

**Project file**
n/a

**Log file**
log of 4.7.1
[cura.log]

**Additional information**
This issue does not occur on cura version pre 4.6.2.
4.6.1 was the first appimage I used on this computer, so I tried 4.6.0 before opening this issue, which also works as expected.
The problem started with 4.6.2 and still exists on 4.7.1
Also I see the cura.cfg file under /home/user/.config/cura/4.7 gets updated.

konskarm, 2020-09-23:
There seems to be an issue with your configuration folder in general. Maybe Cura does not have the necessary privileges and cannot write the application data or cannot create the resources directories.
Could you please share your cura log files? We may be able to find something in there.

WPT-git, 2020-09-23:
That would make sense to me, if Cura 4.6.1 and the new downloaded 4.6.0 would also not save those settings, but they do.
So why does 4.6.2 not work as it is the same 4.6 folder, isn't it?

I will try to attach the log files again, I tried it in the opening post but it seems I accidently deleted the url to it, my bad.

.local/share/cura/4.6 [cura.log](https://github.com/Ultimaker/Cura/files/5268507/cura.log)
.local/share/cura/4.7 [cura.log](https://github.com/Ultimaker/Cura/files/5268519/cura.log)

konskarm, 2020-09-23:
I see that the warning `WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition <definition> failed to pickle.` consistently pops up for all your printers. Looking at [the code that produces this message](https://github.com/Ultimaker/Uranium/blob/167d874397c7a26730dd532f0261dea2b023e809/plugins/LocalContainerProvider/LocalContainerProvider.py#L295), it seems that this is caused by a RecursionError produced while pickling (caching) the files, which leads to Python removing those definitions, going back to a fresh Cura installation.

Maybe this is caused by the definitions inheritted from 4.6 for some reason. Can you try to delete the contents of the 4.7 folder (but not the folder itself)? This way you will start a fresh 4.7 installation without inheriting the 4.6 files. Then you can check if the problem persists.

WPT-git, 2020-09-23:
I see, I never thought of cleaning the folder. Thanks for your solution.

**Indeed it did work**, now I have to reassign all changed values as the profile export from 4.6.1 to 4.7.1 did not really work (visibility settings were imported but the number values are different from the created profile). But it's less bugging to recreate the profiles than the issue I had until now.

I still fail to see why 4.6.2 has the same issue though, as it uses the same folder/cache/files like every other 4.6.* version uses.

What should I do in the future to avoid having this issue once again? Usually I just download the new appimage and use it right away.

nallath, 2020-09-24:
Setting the permissions of the config folders so that Cura is allowed to change them should do the trick.

closed 2024-12-08 by github-actions[bot]