Ultimaker/Cura #8431: Allowing External instances or Add Replace object tool in the right click menu.

JRTrudel, 2020-09-23:
Hi!

I just realize when you save has project, Cura remove the ability to replace the object.    So if you want to update the object you have to redone all custom operation like Support Blockers.  I find it really counter productive and annoying.   We should be able to reopen the project and replace the object.

A other option could be to allow save objects in project has Instances using external path.    Having the possibility to replace an object appear to me really important...  Is it possible to add this option (or Feature)  or at least allow replacing saved Object in the future?

Thank you very much

Best Regards

nallath, 2020-09-24:
You can already do this with the MeshTools plugin (https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools). 

Select the model that you want to replace
Extensions - > MeshTools -> replace -> select file you want it replaced with.

JRTrudel, 2020-09-24:
When I do your repro, the object is not replace and a new one il loaded over it.  After that they are fuse togeter in one entity...

[image]

fieldOfView, 2020-09-24:
See also: https://github.com/fieldOfView/Cura-MeshTools/issues/16

Ghostkeeper, 2020-09-25:
I'm unable to reproduce this issue. It doesn't load them as a group when I open a project file.
[image]

Which version of Cura are you using? Could we see a log file and a project file that you are opening?

JRTrudel, 2020-09-25:
I use version 4.7.0.   Not sure where are the log files?      
https://drive.google.com/file/d/182AufvxrR5l1ENFzwZT8W8yr7li_Z8aZ/view?usp=sharing

I also post in the other Issue  https://github.com/fieldOfView/Cura-MeshTools/issues/16

Maybe we can continue there to avoid duplication...???

Ghostkeeper, 2020-09-28:
Well, the MeshTools may provide a workaround with its reload function, but it should not group models in the first place by saving and reloading a project file. I'd consider that a bug if it does.

Ghostkeeper, 2020-09-28:
Hmm, your file loads as a group, but that's because it was apparently saved as a group. From the 3MF file:

```
<object id="4" name="GarnlyPrimo_CanopyWAntena_AXXI.stl" type="model">
	<components>
		<component objectid="1" transform="1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0" />
		<component objectid="2" transform="3.402359576483805 -1.270494394264317 2.162027827149971 0.945848336967765 0.8591474309796329 0.3703585528848981 -0.5507612726065617 0.1856796156290325 0.9758398280661956 -19.3630990846835 5.236846097649863 -15.234980024126656" />
		<component objectid="3" transform="4.101299486931837 -1.3459303082184872 -1.879564045555594 0.311810312384533 0.9501441237699655 2.6229864608029314e-08 0.3793265573989604 -0.12448423277479291 0.9168505728045562 -20.380357397047458 4.782436483549866 15.55516986988401" />
	</components>
</object>
```

When I ungroup it, then save and re-load that file, it doesn't automatically group them.

Keep in mind that you can still Ctrl+click on a mesh to replace that one specifically.

JRTrudel, 2020-09-28:
I never group the objects...   So it mean it do it by itself.  I just test it.  Was not grouped before saving the project.   And Grouped after reloading it.    Should I report it?

In addition when I ungroup it the Support Block move in space and I can't move them anymore..!?  Why?  

For me the pipeline is broken

Ghostkeeper, 2020-09-29:
I'd like to know what your exact reproduction steps are that causes it to group models when saving/loading a project file. These are the steps that I've tried:

1. Open Cura, or at least have a clear build plate.
2. Load two cubes into the scene.
3. Go to File -> Save Project and save the project file somewhere.
4. Clear the build plate (Ctrl+D).
5. Open the project file.

The result is 2 cubes, not grouped.

The result looks like this: http://dulek.net/work/not_grouping_after_reload_project.webm

It's not grouping anything. Can you repeat my steps? What steps do you use to arrive at your broken result?

JRTrudel, 2020-09-29:
test with this STL
https://drive.google.com/file/d/1Kxh0m1duVGM2hP2L4EEzpgqn6DrIgYE6/view?usp=sharing

Ghostkeeper, 2020-10-02:
Same result with that STL, placing two support blockers on the "wings". What were your exact reproduction steps to produce the 3MF file you shared earlier?

closed 2020-10-16 by no-response[bot]