Ultimaker/Cura #8434: Octoprint plugin not honouring "Select model after uploading"

MaxwellSmartAgent-86, 2020-09-24:
**Application version**
Happens on 4.7.0 and 4.7.1

**Platform**
Windows 10, Intel XEON CPU, 24GB RAM

**Printer**
Creality Ender 3

**Reproduction steps**
1. Uncheck the box Select print job after uploading

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Sending the job to the printer (whilst printing" still results in a warning that the job couldn't be selected because the print is printing.

**Expected results**
The above mentioned warning should not appear if the Select print job after uploading box is unchecked.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I noticed the above happening after I installed Cura 4.7.0 and the behaviour has not changed in Cura 4.7.1.

nallath, 2020-09-24:
@fieldOfView Is this an issue with the plugin or something on our end?

fieldOfView, 2020-09-24:
Sounds like something on my end. @MaxwellSmartAgent-86, could you open an issue at https://github.com/fieldofview/Cura-OctoPrintPlugin/issues ?

closed 2020-09-24 by nallath

fieldOfView, 2020-09-24:
Continued at https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/195