Ultimaker/Cura #8435: NMake does not build all dependencies - Building on Windows

JaideepBgit, 2020-09-24:
**Application Version**
BlackBelt CURA 3.6 version

**Platform**
Processor: Intel(R) Core(TM) i5-8300H CPU @ 2.30 GHz
RAM: 16.0 GB
System type: 64-bit Operating system, x64-based processor

**Printer**
Blackbelt Printer
 
**Steps to Reproduce**
I am trying to run source from [https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#microsoft-visual-studio-2019](url). 
And attempting to run step 6.
6. Open a terminal and run NMake in the build directory. If all goes well, this should build all dependencies.

**Actual Results**
PS D:\work\cura-build-environment-master\build> NMake

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1064: MAKEFILE not found and no target specified
Stop.

**Expected results**
Should build all dependencies.

rburema, 2020-09-25:
Does the alternative workflow work for you?

> If this workflow doesn't work out for you, you can proceed below by installing the dependencies manually.

JaideepBgit, 2020-09-25:
Hi @rburema, I have used CMAKE application and since Visual Studio was installed, it make files with extension of .vcxproj instead of makefiles. So I have used this command:
cmake -DCMAKE_INSTALL_PREFIX=../install_dir -DBUILD_STATIC=ON -DCMAKE_BUILD_TYPE=Release -G "NMake Makefiles" ..
from terminal in build folder and it worked. I created makefile for nmake. 

Then nmake worked.

JaideepBgit, 2020-09-25:
But I have another issue with Savitar when building all dependencies together:

LINK: command "D:\MicrosoftVisualStudio14.0\VC\bin\amd64\link.exe /nologo @CMakeFiles\python_module_Savitar.dir\objects1.rsp /out:Savitar.pyd /implib:_Savitar.lib /pdb:D:\work\cura-build-environment-master1\cura-build-environment-master\build\Savitar-prefix\src\Savitar-build\Savitar.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO C:\Program Files\Python38\libs\python38.lib Savitar.lib pugixml\pugixml.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:Savitar.pyd.manifest" failed (exit code 1104) with the following output:
LINK : fatal error LNK1104: cannot open file 'python35.lib'

So I have tried to use alternate workflow, but got this error:
D:\work\libSavitar-master\libSavitar-master\build>cmake -DCMAKE_INSTALL_PREFIX=../install_dir -DBUILD_STATIC=ON -DCMAKE_BUILD_TYPE=Release -G "NMake Makefiles" ..
-- The C compiler identification is MSVC 18.0.40629.0
-- The CXX compiler identification is MSVC 18.0.40629.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/work/libSavitar-master/libSavitar-master/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake /nologo cmTC_d2dca\fast &&       "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\nmake.exe"  -f CMakeFiles\cmTC_d2dca.dir\build.make /nologo -L                  CMakeFiles\cmTC_d2dca.dir\build
    Building C object CMakeFiles/cmTC_d2dca.dir/testCCompiler.c.obj
        C:\PROGRA~2\MICROS~1.0\VC\bin\cl.exe @C:\Users\Poligy\AppData\Local\Temp\nmA575.tmp
    testCCompiler.c
    Linking C executable cmTC_d2dca.exe
        "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_d2dca.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests -- C:\PROGRA~2\MICROS~1.0\VC\bin\link.exe /nologo @CMakeFiles\cmTC_d2dca.dir\objects1.rsp @C:\Users\Poligy\AppData\Local\Temp\nmA5B4.tmp
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_d2dca.dir/manifest.res CMakeFiles\cmTC_d2dca.dir/manifest.rc" failed (exit code 0) with the following output:
    The system cannot find the file specifiedNMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\nmake.exe"' : return code '0x2'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

-- Configuring incomplete, errors occurred!

JaideepBgit, 2020-09-25:
Hi @rburema, I found the error in the build files that it was pointing to python38 when the cura installed one is python3.5.7. Python38 is the system version. Hence I deleted the system python and ran the build process from start and it worked.

closed 2020-09-25 by JaideepBgit

rburema, 2020-09-25:
@JaideepBgit  Thanks for reporting back! I might clarify a few things in the write-up as a result of this.