Ultimaker/Cura #8437: Two extruders printing outlines

jzagaja, 2020-09-24:
I would like to print advanced laminates. Like this:

two extruders printing defined outline number so that we have:

outline no 1: extruder 1
outline no 2: extruder 2
outline no 3: extruder 1
infill: extruder 1 or 2
outline no 4: extruder 1
outline no 5: extruder 2
outline no 6: extruder 1

konskarm, 2020-09-25:
What is this feature needed for? I am afraid that it will cause problems if the extruders contain materials that do not stick together very well.

jzagaja, 2020-09-25:
Feature is needed to make advanced laminates with constrained layer damping. Perfect stick is not needed.

konskarm, 2020-09-25:
That's a clear use-case. I'll put it up for discussion with the team.

Ghostkeeper, 2020-09-28:
Would you expect it to still print the outlines in order from outside to inside? You can achieve this with infill meshes otherwise, but it'll first do everything with one extruder and then everything with the other extruder, so it'll leave gaps in the walls that will later be filled with the other extruder, which is not ideal I'd think. But if it does need to print in order you'll need to switch multiple times per layer which would make the prime tower a whole lot more complex.

jzagaja, 2020-09-28:
Yes from outside to inside switching extruders.

jzagaja, 2020-10-01:
Coextruded PMMA+PVC is used now by window makers. Why can't we try mix materials in outlines? Allow freedom of choice.

konskarm, 2020-10-01:
Hi @jzagaja. I discussed this with out materials team and the use case is not enough to justify the cost of implementation. It is very specific and we expect that it will not be used widely.

closed 2020-10-01 by konskarm

jzagaja, 2020-10-01:
Ok will ask other slicing soft manufacturers.