Ultimaker/Cura #8438: 3D Print pauses and leaves a big blob

JamesEdwardKillick, 2020-09-24:
**Application version**
Cura v 4.7.1

**Platform**
Windows 10

**Printer**
Prusa i3

**Reproduction steps**
I scaled a print to 170% and repositioned on bed. Started print like I have done many, many times before.

**Screenshot(s)**
No Screenshots

**Actual results**
During the print at 2 specific points I get 105 on the bottom line of the display and the print stops at the end of the line of print. When i press the select button on my printer, it starts again, but if I'm not watching it then i get a big blob in my print. The displasy then shows M105 on the bottom line of the display and stops at the end of the next line of print. This happens again on the second layer.

I am printing with a wood PLA 1.75mm through a 0.5mm nozzle with extruder set to 200 degrees and table set to 60 degrees. My printer is a Geeetech i3b pro. This is not in the list of printers hence the prusa i3 being selected.

**Expected results**
Thge print should run from start to finish without stopping. This pausing with code 105 only happens on some prints. May be coincident with scaling the print away from 100%
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Liger0, 2020-09-24:
A project file and gcode would help the developers I think.
 Does the gcode contain that pause script or is it the firmware interrupting? Maybe a defective filament sensor?

Ghostkeeper, 2020-09-28:
Are you printing via USB cable?

`M105` is the [g-code command to read the current extruder temperature](https://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature). It should not pause the print.

But yes a project file would help us see if there is something that could cause the printer to slow down, and if so, we could see how it got there.

JamesEdwardKillick, 2020-09-28:
Hi, I load an stl file, enlarge to 170% in all directions, then slice. It’s weird that it always stops at the same places. I can send the stl file if it helps...

Sent from my Sonic Screwdriver...

Liger0, 2020-09-28:
stl can't be of any help. At least the gcode is required, better the project itself.

JamesEdwardKillick, 2020-09-28:
I don’t have a project, I got the stls from thingiverse. I can send the gcode made by cura. Would that be o.k.?

Thanks...

Sent from my Sonic Screwdriver...

JamesEdwardKillick, 2020-09-28:
I tried to attach the gcode but then I got a response to say the attached was too big...

Sent from my Sonic Screwdriver...

Liger0, 2020-09-28:
Zip it.

JamesEdwardKillick, 2020-09-28:
zipped gcode attached...

Thanks...

Ghostkeeper, 2020-09-29:
> I don’t have a project, I got the stls from thingiverse.

There is a "Cura project file" format which Cura can save, that includes the build plate including all your transformations, and your current profiles and settings. It allows us to not only see the g-code output you're getting, but also how it got there. This is why we include it in our template.
You can create a project file by loading in your model in Cura and going to File -> Save Project...
The resulting file has to be zipped as well in order for Github to accept it.

> zipped gcode attached...

This website (Github) doesn't accept attachments to e-mail replies. You'll have to go to [your issue page](https://github.com/Ultimaker/Cura/issues/8438) and reply there manually, attaching a file to your reply.

closed 2020-10-14 by no-response[bot]