Ultimaker/Cura #8439: Publish to Windows Package Manager

denelon, 2020-09-24:
I'd like to have releases added to the Windows Package Manager automatically.

**Describe the solution you'd like**
When a new version is built, generate a manifest and submit a PR to the Microsoft Community Package manager repository.
https://github.com/microsoft/winget-pkgs

**Affected users and/or printers**
All Windows 10 users will have the Windows Package Manager when it is generally available. All Windows Insiders, Windows Package Manager Insiders, and users who have downloaded and installed the Windows Package Manager from GitHub (https://github.com/microsoft/winget-cli) have the capability today.

**Additional context**
The first manifest for Ultimaker.Cura was generated on May 14 (https://github.com/microsoft/winget-pkgs/blob/master/manifests/Ultimaker/Cura/4.5.0.yaml) before the preview was announced (https://devblogs.microsoft.com/commandline/windows-package-manager-preview/). Instructions to join the preview are at the bottom of the most recent blog post (https://devblogs.microsoft.com/commandline/windows-package-manager-preview-v0-2-2521/).

`winget install cura` will download the installer, validate the SHA256 and install the latest version of the package in the repository.

konskarm, 2020-09-25:
Hi @denelon. We discussed this internally we will skip that for the time being.

denelon, 2021-05-15:
@konskarm, thanks for the update :smile:. 

I just added the [manifest](https://github.com/microsoft/winget-pkgs/pull/13020) for version 4.9.1. I'm using the .exe installer which supports "side-by-side" installation. I noticed some comments in the repository regarding an .msi installer. Generally, on [Windows](https://docs.microsoft.com/en-us/windows/msix/supported-platforms), the [MSIX](https://docs.microsoft.com/en-us/windows/msix/overview) packaging format is now preferred. 

Should I make a separate Issue for that request?

I'd love to see if there are ways to preserve Cura configurations using the newer app package format.

Ghostkeeper, 2021-05-31:
> I noticed some comments in the repository regarding an .msi installer. Generally, on [Windows](https://docs.microsoft.com/en-us/windows/msix/supported-platforms), the [...]

We provide an MSI installer for Cura Enterprise, the version of Cura which only allows security-reviewed plug-ins to be installed. The MSI installer is less fully-featured, but does allow automatic headless installation, which is something that IT administrators want in order to deploy Cura on computers in their business network. The MSI installer doesn't allow multiple versions of Cura to be installed at the same time, and is generally more of an ass when it comes to removing old configuration, so it's not an option for us to distribute our normal releases that way.

denelon, 2021-06-01:
@Ghostkeeper, thank you for the context. Let me now if you would like me to reach out to someone here at Microsoft to see if there are some "best practices" or alternatives to try and upgrade/preserve configurations and address the install directory challenge.

rbarbrow, 2021-10-18:
It is still set to 4.9 any word on when this will be completed?

Ghostkeeper, 2022-04-02:
Automation is helpful, but it's naive to think that the automation doesn't require maintenance as well. Especially when it concerns something like using a Microsoft platform which is subject to change at any time.

vedantmgoyal2009, 2023-06-30:
You can use winget-releaser github action

halr9000, 2023-07-05:
Please revive this issue! Would love to check out the new tree supports and smart brim! ;) 

```
Ultimaker Cura Ultimaker.Cura 5.3.1   winget
PS C:\Users\hal> winget search ultimaker --source winget
Name           Id             Version
-------------------------------------
Ultimaker Cura Ultimaker.Cura 5.3.1
```

But, alas, I failed in my efforts.