Ultimaker/Cura #8442: 4.7.1 - Some of the support merges with print

wyekey, 2020-09-25:
Application version - 4.7.1
Platform - Windows 10 x64 
Printer - Ender 3

Issue has occurred on every print since I upgraded from 4.7 to 4.7.1. I have printed five models, four were downloaded from thingiverse and myminifactory, and one was a model I designed and printed successfully previously ( prior to Cura version 4.7.1 ). Parts of the support filament merge with the print, both when the support is from the bed upwards and when coming off a part of the print itself.

**Reproduction steps**
Import into Cura - slice - save file - print.

**Screenshot(s)**
For my design, the model shown in Cura prior to slicing is correct. 

1) The design has four bolt holes and a slightly curved base ( it fits over the crossbar of my bicycle ). Filament is printed from the bed upwards for support, which should just peel off and leave a smooth curved surface. 

[image]

the model upside down to show the curved surface better

[image]
  
[image]

2) There are two small rectangular holes in the side of the print that go right through and out the other side.  

[image]

**Actual results**

1) The only support you can peel off are four small domes around the holes, and two small end parts, so it's very uneven and looks like four craters. The rest of the support material is part of the print ( no, it's not material I just haven't peeled away, it is physically part of the print ).

[image]
[image]

2) The holes on the side of the model are filled in. Whilst not that clear, I've held the print against my bright monitor, and you should be able to see a small hole.

[image]

**Additional information**
These issues can be seen in the sliced model in Cura. 

Attached are a screenshot of the support settings I have changed from the default and relevant log files requested.
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/5279818/cura.log)

I have the print as an stl file and 3mf file, but your system won't accept them.

Asterchades, 2020-09-25:
This is a setting issue, though I'm unsure whether it's a default (I suspect not) or something you changed yourself and forgot about (most likely).

In your log is the setting 'conical_overhang_enabled="True"'. Within Cura, this becomes the more human-readable "Make Overhang Printable" experimental setting. And it would appear that it's doing exactly what it's supposed to do, in that the cones its printing are automatically being angled down at 45° (your specified "Support Overhang Angle") to make it so they can print successfully.

This has nothing to do with Supports, which is why you've not seen it mentioned before. It's strictly that experimental setting, and it would have printed identically if you'd not used Supports at all. This same setting would have been affecting all prints you've been doing with overhangs greater than 45° or bridges, though it may not have been quite as noticeable as it was here..

As for the directories, minor builds don't get a new director for either program or setting. As such 4.7.1 goes into the same 4.7 directory, as would 4.7.2 if it ever came into being.

wyekey, 2020-09-25:
Unticked the "make overhang printable" option and it slices correctly now. Thanks. Love Cura by the way.

closed 2020-09-25 by wyekey