Ultimaker/Cura #8443: No extrude first perimeter at layer 17 at this model

krnelpanic, 2020-09-25:
Cura 4.7.1

**Platform**
Windows 10 x64 i7 9700k Nvidia RTX 2070

**Printer**
Anet ET4

**Reproduction steps**
When I print this [](https://drive.google.com/file/d/18Ld27b1idUP2OSs43pg6Lq6E2ytcPhuA/view?usp=sharing) at layer >17 doesn't extrude first 2 perimeters of the left hole. I did this print with Slic3r without problems.

Layer eight 0.25 nozzle 0.4 perimeters 3 infill 15 gyroid

**Actual results**
[](https://ibb.co/qJ6McMt)

Ghostkeeper, 2020-09-28:
From what I can see it prints a perimeter fine at layer 17:
[image]

The g-code also instructs the printer to print these perimeters:

```
;TYPE:WALL-INNER
;MESH:Funky Crift-Jaban (1).stl
G1 F1500 E1258.92913
G1 X129.903 Y113.103 E1258.94187
G1 X130.208 Y113.089 E1258.95456
G1 X130.508 Y113.043 E1258.96718
G1 X130.803 Y112.97 E1258.97981
G1 X131.091 Y112.867 E1258.99253
G1 X131.364 Y112.737 E1259.0051
G1 X131.626 Y112.581 E1259.01778
G1 X131.871 Y112.4 E1259.03044
G1 X132.097 Y112.195 E1259.04313
G1 X132.302 Y111.968 E1259.05584
G1 X132.482 Y111.723 E1259.06848
G1 X132.639 Y111.465 E1259.08104
G1 X132.77 Y111.187 E1259.09382
G1 X132.872 Y110.901 E1259.10644
G1 X132.946 Y110.605 E1259.11913
G1 X132.991 Y110.304 E1259.13178
G1 X133.005 Y110 E1259.14443
G1 X132.991 Y109.696 E1259.15708
G1 X132.946 Y109.395 E1259.16974
G1 X132.872 Y109.1 E1259.18238
G1 X132.77 Y108.813 E1259.19504
G1 X132.639 Y108.536 E1259.20778
G1 X132.482 Y108.275 E1259.22045
G1 X132.302 Y108.032 E1259.23302
G1 X132.097 Y107.806 E1259.24571
G1 X131.871 Y107.601 E1259.25839
G1 X131.627 Y107.42 E1259.27102
G1 X131.365 Y107.263 E1259.28372
G1 X131.091 Y107.133 E1259.29633
G1 X130.803 Y107.03 E1259.30905
G1 X130.507 Y106.957 E1259.32172
G1 X130.207 Y106.912 E1259.33433
G1 X129.903 Y106.898 E1259.34698
G1 X129.598 Y106.912 E1259.35968
G1 X129.299 Y106.957 E1259.37225
G1 X129.002 Y107.029 E1259.38495
G1 X128.714 Y107.133 E1259.39769
G1 X128.44 Y107.263 E1259.41029
G1 X128.178 Y107.42 E1259.42299
G1 X127.934 Y107.601 E1259.43562
G1 X127.708 Y107.806 E1259.44831
G1 X127.503 Y108.033 E1259.46103
G1 X127.323 Y108.277 E1259.47363
G1 X127.166 Y108.537 E1259.48626
G1 X127.036 Y108.812 E1259.49891
G1 X126.933 Y109.1 E1259.51162
G1 X126.859 Y109.395 E1259.52427
G1 X126.814 Y109.696 E1259.53692
G1 X126.8 Y110 E1259.54957
G1 X126.814 Y110.304 E1259.56222
G1 X126.859 Y110.605 E1259.57488
G1 X126.933 Y110.901 E1259.58756
G1 X127.036 Y111.189 E1259.60028
G1 X127.166 Y111.463 E1259.61289
G1 X127.322 Y111.723 E1259.62549
G1 X127.503 Y111.968 E1259.63816
G1 X127.708 Y112.195 E1259.65087
G1 X127.934 Y112.4 E1259.66356
G1 X128.179 Y112.581 E1259.67622
G1 X128.441 Y112.737 E1259.6889
G1 X128.714 Y112.867 E1259.70147
[... 193 lines cut ...]
G1 X87.372 Y109.473 E1261.75168
G1 X87.333 Y109.735 E1261.76269
G1 X87.32 Y110 E1261.77372
G1 X87.333 Y110.265 E1261.78476
G1 X87.372 Y110.527 E1261.79577
G1 X87.436 Y110.785 E1261.80682
G1 X87.525 Y111.034 E1261.81781
G1 X87.639 Y111.275 E1261.8289
G1 X87.776 Y111.501 E1261.83988
G1 X87.933 Y111.715 E1261.85092
G1 X88.111 Y111.912 E1261.86196
G1 X88.308 Y112.09 E1261.873
G1 X88.521 Y112.248 E1261.88402
G1 X88.749 Y112.384 E1261.89506
G1 X88.988 Y112.497 E1261.90605
G1 X89.238 Y112.587 E1261.9171
G1 X89.495 Y112.65 E1261.9281
G1 X89.757 Y112.69 E1261.93912
G1 X90.022 Y112.703 E1261.95015
G1 X90.288 Y112.69 E1261.96122
```

Is your printer not executing those g-code commands? What's really happening there?

Perhaps you have a project file for us? You can create one by going to File -> Save... It would really help to fill in the bug report template to debug this sort of issue.

krnelpanic, 2020-09-29:
I can see that will extrude at simulation but not when printing, here is the .gcode and project file
https://drive.google.com/file/d/1IPdLLCDykfvUsRVZ_CqGAjkKELlKa9Hs/view?usp=sharing
https://drive.google.com/file/d/1FmEdmxU9WttxnxFRssUR1u1Z9gOhC5GR/view?usp=sharing

Ghostkeeper, 2020-10-02:
In your g-code I can see the instructions clearly that print all three perimeters for all 3 holes on layer 17 (as well as each other layer):

```
;TYPE:WALL-INNER
G1 F3000 E1097.61252
G1 F2400 X92.936 Y111.945 E1097.6275
G1 X93.112 Y111.652 E1097.64242
G1 X93.259 Y111.342 E1097.6574
G1 X93.375 Y111.017 E1097.67246
G1 X93.459 Y110.683 E1097.68749
G1 X93.51 Y110.343 E1097.7025
G1 X93.525 Y110 E1097.71749
G1 X93.51 Y109.657 E1097.73248
G1 X93.459 Y109.317 E1097.74749
G1 X93.375 Y108.984 E1097.76248
G1 X93.259 Y108.659 E1097.77754
G1 X93.112 Y108.348 E1097.79256
G1 X92.935 Y108.054 E1097.80754
G1 X92.732 Y107.78 E1097.82243
G1 X92.501 Y107.523 E1097.83751
G1 X92.245 Y107.291 E1097.85259
G1 X91.969 Y107.087 E1097.86758
G1 X91.674 Y106.91 E1097.88259
G1 X91.354 Y106.759 E1097.89804
G1 X91.041 Y106.647 E1097.91255
G1 X90.704 Y106.564 E1097.9277
G1 X90.366 Y106.513 E1097.94263
G1 X90.038 Y106.498 E1097.95696
G1 X89.941 Y106.499 E1097.96119
G1 X89.675 Y106.515 E1097.97283
G1 X89.341 Y106.564 E1097.98756
G1 X89.006 Y106.647 E1098.00263
G1 X88.681 Y106.763 E1098.01769
G1 X88.371 Y106.91 E1098.03267
G1 X88.076 Y107.087 E1098.04769
G1 X87.8 Y107.291 E1098.06267
G1 X87.545 Y107.523 E1098.07772
G1 X87.313 Y107.778 E1098.09277
G1 X87.11 Y108.052 E1098.10766
G1 X86.933 Y108.347 E1098.12267
G1 X86.785 Y108.66 E1098.13779
G1 X86.67 Y108.984 E1098.1528
G1 X86.586 Y109.317 E1098.16779
G1 X86.535 Y109.657 E1098.1828
G1 X86.52 Y110 E1098.19778
G1 X86.535 Y110.343 E1098.21277
G1 X86.586 Y110.683 E1098.22778
G1 X86.67 Y111.017 E1098.24281
G1 X86.785 Y111.34 E1098.25778
G1 X86.933 Y111.655 E1098.27298
G1 X87.11 Y111.946 E1098.28784
G1 X87.313 Y112.221 E1098.30277
G1 X87.544 Y112.478 E1098.31785
G1 X87.8 Y112.71 E1098.33293
G1 X88.077 Y112.915 E1098.34798
G1 X88.372 Y113.09 E1098.36295
G1 X88.68 Y113.237 E1098.37785
G1 X89.006 Y113.353 E1098.39295
G1 X89.025 Y113.357 E1098.3938
G1 X89.339 Y113.436 E1098.40793
G1 X89.677 Y113.486 E1098.42285
G1 X90.005 Y113.503 E1098.43719
G1 X90.104 Y113.502 E1098.44151
G1 X90.373 Y113.486 E1098.45327
G1 X90.706 Y113.436 E1098.46797
G1 X91.026 Y113.357 E1098.48236
G1 X91.12 Y113.326 E1098.48668
G1 X91.366 Y113.235 E1098.49813
G1 X91.672 Y113.09 E1098.51291
G1 X91.968 Y112.915 E1098.52793
[... 371 lines cut ...]
G1 X127.655 Y111.501 E1113.61999
G1 X127.813 Y111.715 E1113.6316
G1 X127.991 Y111.912 E1113.64319
G1 X128.188 Y112.09 E1113.65478
G1 X128.401 Y112.248 E1113.66636
G1 X128.629 Y112.384 E1113.67795
G1 X128.868 Y112.497 E1113.68949
G1 X129.118 Y112.587 E1113.70109
G1 X129.376 Y112.65 E1113.71268
G1 X129.637 Y112.69 E1113.72421
G1 X129.903 Y112.703 E1113.73583
G1 X130.168 Y112.69 E1113.74741
G1 X130.43 Y112.65 E1113.75898
G1 X130.685 Y112.587 E1113.77045
G1 X130.737 Y112.57 E1113.77284
G1 X130.937 Y112.497 E1113.78213
G1 X131.176 Y112.384 E1113.79367
G1 X131.404 Y112.248 E1113.80526
G1 X131.617 Y112.09 E1113.81684
```
[image]

You may just be having flow problems. Is your nozzle unclogged? These moves are preceded by infill with a lot of travel moves. You could consider enabling Connect Infill Lines for a more consistent flow, or selecting a different infill pattern.

As far as I can see, the output of Cura is correct. Your settings may not be.

krnelpanic, 2020-10-02:
Maybe doesn't extrude enought?

I did 18 attemps with cura modifing some parameters with similar result, just 1 with PrusaSlicer with good result
I didn't adjust anything at printer just sliced same model at diferent app
Did flow test with same result at both apps
Nozzle and PLA was the same at all prints

[image]

Ghostkeeper, 2020-10-05:
From the g-code you can pretty easily calculate whether the amount of extrusion is correct. Take for instance these two commands from the snippet I posted above:

```
;TYPE:WALL-OUTER
G1 F2400 X124.603 Y104.453 E1108.23009
G1 X95.397 Y104.453 E1109.50505
```

It first moves from whatever its last position was to `(124.603, 104.453)`, and then moves to `(95.397, 104.453)`. In this second move it moves the extruder from `1108.23009` to `1109.50505`.

So it's making a move with a length of [29.206mm](https://www.wolframalpha.com/input/?i=124.603+-+95.397) and in that space it extrudes [1.27496mm](https://www.wolframalpha.com/input/?i=1109.50505+-+1108.23009) of filament. With a filament diameter of 1.75mm, that filament length amounts to [3.067mm³](https://www.wolframalpha.com/input/?i=%281109.50505mm+-+1108.23009mm%29+*+pi+*+%281.75mm%2F2%29%5E2) in volume. Dividing that by the length (29.206mm) and thickness (0.25mm) of the line, results in a line width of [0.42mm](https://www.wolframalpha.com/input/?i=%281109.50505mm+-+1108.23009mm%29+*+pi+*+%281.75mm%2F2%29%5E2+%2F+29.206mm+%2F+0.25mm). So yes, from the g-code, it appears that the amount of extrusion is correct.

Of course it could still be slipping in the feeder. Or oozing on the way to the wall. Or there could be a spring-tension in the filament between the feeder and the nozzle. There are an infinite amount of subtle differences between any two slices so the comparison with any other slicer is moot. We really just need to know what you'd like to have different in your g-code and why. We can't print with your printer so there is no other way in which we can help you optimize your print.

closed 2020-10-19 by no-response[bot]