Ultimaker/Cura #8444: Cura 4.7 and 4.7.1 crash at preview for large model

war4peace, 2020-09-25:
**Application version**
4.7.0, 4.7.1, possibly earlier.

**Platform**
Windows 10 x64 1909
2x GTX 1080 Ti in SLI
CPU: Intel i7 6800K
RAM 32GB DDR4 @3000 MHz

**Printer**
Creality CR 10 MAX

**Reproduction steps**
I have been commissioned to print a large model. [This is the model](https://www.myminifactory.com/object/3d-print-alexander-and-diogenes-133982) and it needed to be resized such as the largest size should be 420mm.

After resizing and arranging the model on the plate like a litophane, I started slicing. Sometimes Cura would not slice at all immediately saying `"Unable to slice", after moving the model back and forth by 1mm on the plate I try again and eventually it starts slicing, which takes a while.

After it finishes, I can't preview the model. Cura crashes every time I try doing that. Below you can read the relevant log.
Interestingly, I get a warning saying `"Mesh has overlapping faces!" - I checked with various tools (Netfabb, meshMixer) and the model is manifold. At any rate, when I try to preview the sliced G-Code (it's 1.05 GB in size), Cura memory usage jumps to around 8-9 GB, fluctuates there and crashes. Logs show no crash reason, but I assume there is a memory overflow exception somewhere triggered by this large and complex model.

With Cura being the only slicer reliably using Tree Supports, and this being a complex model which doesn't really work otherwise, I am stuck.

FYI I have successfully sliced and previewed the model at 100% scale and printed it with no issues.

**Project file**
The generated GCode is 1.05 GB large. Here are the steps to reproduce:
Download this model:
https://www.myminifactory.com/object/3d-print-alexander-and-diogenes-133982
Rotate the model on the plate in such a way that the baso-relief stands upright.
Resize model in Cura to Z=420 (all other dymensions synchronized)
Enable Tree Supports
Slice the model.

**Log file**
`2020-09-21 08:05:04,009 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [246]: Starting to slice...
2020-09-21 08:05:04,010 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [264]: Going to slice build plate [0]!
2020-09-21 08:05:04,465 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [457]: Sending slice message took 0.4547562599182129 seconds
2020-09-21 08:05:06,572 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] Mesh has overlapping faces!
2020-09-21 08:05:07,616 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING]  -s wireframe_roof_outer_delay="0.2" -s support_conical_enabled="False" -s switch_extruder_extra_prime_amount="0" -s support_roof_line_width="0.4" -s speed_topbottom="25.0" -s z_seam_y="450" -s wireframe_nozzle_clearance="1" -s wireframe_printspeed_flat="5" -s support_infill_rate="0" -s ironing_pattern="zigzag" -s wireframe_top_jump="0.6" -s cutting_mesh="False" -s wireframe_printspeed="5" -s support_top_distance="0.2" -s acceleration_topbottom="500" -s skin_outline_count="1" -s jerk_wall_0="8" -s material_flow_dependent_temperature="False" -s raft_smoothing="5" -s machine_max_feedrate_z="10" -s machine_always_write_active_tool="False" -s wipe_retraction_retract_speed="25" -s machine_steps_per_mm_z="50" -s support_extruder_nr="0" -s xy_offset="0" -s material_break_preparation_retracted_position="-16" -s prime_tower_min_volume="6" -s layer_start_x="0.0" -s raft_interface_acceleration="500" -s optimize_wall_printing_order="True" -s bridge_skin_material_flow_3="110" -s acceleration_wall_0="500" -s speed_support_infill="25.0" -s support_enable="True" -s raft_surface_line_width="0.4" -s material_flush_purge_length="60" -s wireframe_flow_connection="100" -s line_width="0.4" -s flow_rate_max_extrusion_offset="0" -s support_skip_zag_per_mm="20" -s wall_line_width_x="0.4" -s support_bottom_offset="0.0" -s material_break_temperature="50" -s roofing_material_flow="100" -s travel_avoid_other_parts="True" -s material_bed_temperature_layer_0="60" -s cool_fan_full_layer="4" -s infill_enable_travel_optimization="False" -s acceleration_print="500" -s machine_max_acceleration_z="100" -s travel_compensate_overlapping_walls_enabled="True" -s wall_0_extruder_nr="-1" -s adaptive_layer_height_enabled="False" -s small_feature_speed_factor="50" -s speed_wall_x="25.0" -s outer_inset_first="False" -s cool_fan_speed_min="100.0" -s acceleration_support="500" -s acceleration_wall_x="500" -s draft_shield_height_limitation="full" -s wall_min_flow="0" -s raft_base_speed="18.75" -s support_roof_material_flow="100" -s material_break_speed="25" -s retract_at_layer_change="False" -s support_bottom_line_distance="2.4000240002400024" -s support_type="everywhere" -s infill_material_flow="100" -s machine_firmware_retract="False" -s infill_support_angle="40" -s minimum_bottom_area="10" -s meshfix_keep_open_polygons="False" -s meshfix_maximum_travel_resolution="0.05" -s retraction_speed="25" -s wireframe_printspeed_up="5" -s raft_interface_speed="18.75" -s wall_line_width_0="0.4" -s wireframe_printspeed_bottom="5" -s wipe_retraction_enable="True" -s magic_fuzzy_skin_point_dist="0.8" -s machine_settings="0" -s wall_line_count="2" -s machine_extruder_count="1" -s infill_offset_y="0" -s infill_multiplier="1" -s machine_steps_per_mm_x="50" -s raft_airgap="0.3" -s bridge_skin_material_flow="60" -s support_bottom_material_flow="100" -s print_sequence="all_at_once" -s support_xy_distance_overhang="0.4" -s wireframe_fall_down="0.5" -s speed_travel="150.0" -s support_bottom_stair_step_min_slope="10.0" -s machine_heated_build_volume="False" -s support_angle="45" -s jerk_travel="8" -s support_initial_layer_line_distance="0" -s support_roof_extruder_nr="0" -s infill_wall_line_count="0" -s acceleration_support_infill="500" -s wireframe_top_delay="0" -s support_pattern="zigzag" -s cooling="0" -s coasting_enable="False" -s support_infill_sparse_thickness="0.2" -s support_tower_roof_angle="65" -s support_fan_enable="False" -s infill_sparse_thickness="0.2" -s minimum_support_area="2" -s draft_shield_height="10" -s material_standby_temperature="150" -s support_roof_density="33.333" -s wireframe_roof_fall_down="2" -s retraction_hop_only_when_collides="False" -s platform_adhesion="0" -s z_seam_relative="False" -s material_crystallinity="False" -s skirt_brim_speed="20.0" -s support_brim_enable="True" -s acceleration_roofing="500" -s retraction_hop_enabled="False" -s infill_support_enabled="False" -s support_extruder_nr_layer_0="0" -s cool_min_layer_time_fan_speed_max="10" -s wireframe_printspeed_down="5" -s speed_equalize_flow_enabled="False" -s sub_div_rad_add="0.4" -s time="08:05:04" -s material="0" -s expand_skins_expand_distance="0.8" -s top_layers="4" -s wipe_repeat_count="5" -s bridge_fan_speed_3="0" -s support_skip_some_zags="False" -s gradual_support_infill_step_height="1" -s wipe_hop_amount="0.2" -s gradual_infill_steps="0" -s raft_interface_jerk="8" -s switch_extruder_prime_speed="20" -s material_end_of_filament_purge_speed="0.5" -s bridge_skin_speed_2="12.5" -s minimum_roof_area="10" -s brim_replaces_support="True" -s speed_print="50.0" -s support_z_distance="0.2" -s connect_skin_polygons="False" -s machine_max_jerk_e="5" -s wireframe_straight_before_down="20" -s top_bottom_extruder_nr="-1" -s anti_overhang_mesh="False" -s jerk_print_layer_0="8" -s speed_equalize_flow_max="150" -s support_infill_extruder_nr="0" -s speed_print_layer_0="20.0" -s machine_shape="rectangular" -s support_line_distance="0" -s wipe_retraction_amount="6.5" -s jerk_support_bottom="8" -s raft_surface_jerk="8" -s speed_z_hop="5" -s cross_support_density_image="" -s switch_extruder_retraction_amount="16" -s hole_xy_offset="0" -s bridge_skin_speed_3="12.5" -s machine_max_jerk_xy="10" -s support_conical_angle="30" -s gantry_height="30" -s support_interface_skip_height="0.2" -s skin_angles="[ ]" -s initial_layer_line_width_factor="100.0" -s machine_name="Creality CR-10 Max" -s day="Mon" -s travel_compensate_overlapping_walls_x_enabled="True" -s jerk_topbottom="8" -s infill_mesh_order="0" -s cool_min_speed="10" -s machine_min_cool_heat_time_window="50.0" -s prime_tower_size="20" -s jerk_prime_tower="8" -s support_roof_pattern="grid" -s adaptive_layer_height_threshold="0.2" -s raft_speed="25.0" -s raft_surface_fan_speed="0" -s initial_extruder_nr="0" -s machine_steps_per_mm_e="1600" -s retraction_prime_speed="25" -s acceleration_enabled="False" -s z_seam_type="back" -s skirt_brim_line_width="0.4" -s machine_s [... cut ...]
2020-09-21 08:45:34,083 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\toolhandle.shader]...
2020-09-21 08:45:34,099 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\color.shader]...
2020-09-21 08:45:44,225 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-09-21 08:46:23,692 - DEBUG - [Thread-7] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 50.547261238098145 seconds`

Ghostkeeper, 2020-09-28:
...So this took about 3.5 hours to slice. In my case the resulting file size is 248MB. It was able to load the preview fine using my RTX 2070 GPU:
[image]

The resulting g-code contains 6987764 `G1` commands and 839452 `G0` commands. Each extrusion move gets translated to 26 vertices in the 3D layer view, and each travel move to 13 vertices. So for the layer view after the geometry shader you'd need at least [5.4GB of video RAM](https://www.wolframalpha.com/input/?i=%286987764+*+26+%2B+839452+*+13%29+*+%283+*+8+%2B+4%29+bytes). It'll also temporarily need to keep the original vertices in memory while it's working on the geometry shading, resulting in a total of [5.6GB](https://www.wolframalpha.com/input/?i=%286987764+*+26+%2B+839452+*+13%29+*+%283+*+8+%2B+4%29+bytes+%2B+%286987764+%2B+839452%29+*+%283+*+8+%2B+4%29+bytes) spent on the g-code. Cura will also keep rendering the original model in shadow form, and various other things such as the build plate mesh, several render buffers (including the layer for the g-code output)... And then there's Qt and the operating system itself too. Your GPU has 6GB of VRAM. I can imagine that it simply exceeds your available RAM.

I suggest you switch to compatibility view. It doesn't look as fancy, but it also doesn't use nearly as much VRAM. It's an option in the preferences.

Working to the limits of the VRAM capabilities of the GPU is out of scope for us at the moment. The compatibility mode will have to do.

war4peace, 2020-09-28:
The GTX 1080 Ti has 11GB VRAM. Not sure where the 6GB VRAN value came from, but it's not correct.

Ghostkeeper, 2020-09-29:
Ah, yeah I took it from [here](https://www.nvidia.com/en-sg/geforce/products/10series/geforce-gtx-1080-ti/) but you're right, I looked in the wrong column.

Ghostkeeper, 2020-09-29:
Does it help to set layer view to compatibility mode?

Maybe a dxdiag report would help to debug what happened. It could still be memory limits if you are running other applications that also stay in VRAM.

war4peace, 2020-09-29:
DxDiag info: https://drive.google.com/file/d/1Uiy--tVVU698CbIGuVtElRpjAXxgFb_O/view?usp=sharing
When slicing, I have no other VRAM-intensive application. With Cura open and slicing, the total usage is about 2150 MB out of 11 GB. I will keep GPU-Z open to monitor VRAM usage and then provide the data. Slicing takes a while, and sometimes doesn't finish or errors out so I'll add another entry when done.

war4peace, 2020-09-29:
OK, slicing finished. Same crash. Maximum GPU VRAM usage was 4717 MB right before the crash, with compatibility setting enabled. Cura processes the preview right up to 99%, sits there for a few seconds, Cura DRAM usage (computer memory) spikes to values around 8.5 to 9 GB while layer preview is generated, and then the crash happens.

Here is a screenshot of GPU values as well as DRAM values during slicing: http://gpuz.techpowerup.com/20/09/29/jhh.png

Maximum System memory usage was 18449 MB at the time of the crash, out of 32 GB available.
The machine does not run out of resources.

Ghostkeeper, 2020-10-02:
This is the relevant part of that DxDiag:

```
Windows Error Reporting:
+++ WER0 +++:
Fault bucket 1557513878974120893, type 5
Event Name: RADAR_PRE_LEAK_64
Response: Not available
Cab Id: 0

Problem signature:
P1: Cura.exe
P2: 0.0.0.0
P3: 10.0.18363.2.0.0
P4: 
P5: 
P6: 
P7: 
P8: 
P9: 
P10: 
```

This error code `RADAR_PRE_LEAK_64` indicates that Windows' [Resource Exhaustion Detection and Resolution](https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2008-R2-and-2008/cc774730(v=ws.10)?redirectedfrom=MSDN) (with the slightly inaccurate acronym RADAR) has closed down the application because it sees that system resources are getting dangerously low. The RAM and VRAM usage may have been fine before the crash, but the crash may have been due to allocating an additional big chunk of memory. It'll crash before actually allocating that memory. The DxDiag file indicates that your computer is running out of resources.

war4peace, 2020-10-02:
Very interesting.
I have found entries in Event Viewer detailing the same thing, with Exception code: 0xc0000005, meaning STATUS_ACCESS_VIOLATION, meaning Cura tried to access memory it doesn't have access to.
I have been monitoring Cura preview for (relatively) smaller models, e.g. [this one](https://www.thingiverse.com/thing:4369430), and I have noticed that during sliced model preview, Cura has a boost in memory allocation, where it more than doubles in memory consumption for a brief moment, before releasing part of it. So it's indeed possible that the 9+ GB used could have attempted to balloon to 20+, which would indeed have been a problem.

I guess I'll have to switch to 64 GB RAM soon 😄

closed 2020-10-02 by war4peace