Ultimaker/Cura #8446: Dual layer heights

7ucky7uk3, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
When lowering the layer height from 0.2mm to 0.1mm for a higher resolution the print time doubles.

**Describe the solution you'd like**
When activating the "dual layer heights"-feature cura slices the outside wall with 50% of the set layer height. Inner walls, infill and
supports are sliced with 100% layer height. This would keep the print time low while the outside has a fine layer structure.
For example: 
- selected layer height:       *0.2mm*
- printed layer height outside wall:       *0.1mm*
- printed layer height inner walls, infill & supports:       *0.2mm*

**Describe alternatives you've considered**
haven't found any

**Affected users and/or printers**
Everybody

**Additional context**
[image]

konskarm, 2020-09-25:
Hi @7ucky7uk3. While this could save printing time, in practice it's not very feasible. If the entire wall has an angle, the above solution is not possible, since the wall lines of consecutive layers are not aligned on each other. Also, even if they are aligned, the nozzle can bump on the outer wall when trying to print the inner wall or vise-versa (depending on which wall is printed first). 
In general, this whole idea cannot work with our current implementation of CuraEngine, since the slicing is always planar.

7ucky7uk3, 2020-09-26:
Hi @konskarm, thanks for your reply. If the entire wall has an angle, which in most cases is very likely, then the slicer would just have to print 2 or more outer walls (where necessary) to adapt.
But since with the current Cura Engine makes this type of print time optimization impossible, I guess this option has to be put on hold.

closed 2020-09-26 by 7ucky7uk3

Ghostkeeper, 2020-09-28:
See also https://github.com/Ultimaker/Cura/issues/6322 for a different discussion of the same feature.