Ultimaker/Cura #8447: Print preview material length

Tedsond, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
I want to be able to see more specifically how much material will be extruded due to small prints.

**Describe the solution you'd like**
I would like to see that material weight preview could be changed to either extra decimals or setting to mm or cm. 

**Affected users and/or printers**
Anybody who does small prints / Ultimakers

Ghostkeeper, 2020-09-28:
Current precision is down to grams, right? Is that not sufficient?

Tedsond, 2020-09-28:
We do a bridge construction project here at school, which requires bridges to be >2,00g and <4,00g. I want to determine before printing, if the bridge is valid or not. now I can calculate in full grams, I am down to either 1,5g or 2,5g. It's not accurate enough. 

The printer knows quite accurately how much it will use. I would like to see the option and try it out. I will take the 5-10% for granted. if it says a print will be 2,00g and it turns out 2,04. that's not an issue. 

If material length would be in mm instead of meters, I would be able to calculate more accurately myself (I try to do it now aswell, but it gives very different values: between 0,072 and 0,078 gram per cm).

Ghostkeeper, 2020-10-02:
We've discussed this with the development team and decided that it would be good to increase the precision for small prints (e.g. a minimum of 3 significands) but it won't get priority for now. If someone wants to build/adjust that we'll accept it but for now it's deferred until priorities change.

As a workaround, you can also find the material usage to greater precision in the g-code headers, e.g. this print here:

```
;FLAVOR:Marlin
;TIME:7265
;Filament used: 1.34122m ; <-- Using 1.34122m of filament!
;Layer height: 0.1
;MINX:16.302
;MINY:34.065
...
```
You need to convert it yourself though in that case, which requires a bit of computation. The way that Cura does this is as follows:

    material_weight = filament_length_used * pi * (filament_diameter / 2)^2 * filament_density

So for instance with a filament diameter of 2.850mm and using PLA with a density of 1.300g/cm³ in the above print, you'd get `material_weight = 1341.22mm * pi * (2.850mm / 2)² * 0.0013g/mm³ = 11.12g`. The significant digits of this according to scientific notation would be limited by the accuracy of the filament diameter (2.850), the filament density (1.300) and the material length. This material length is written in the g-code header down to 100ths of a millimeter so it's pretty precise.

The Griffin g-code header only writes down material volume in mm³ integers, so that's less accurate. If you're using that, I suggest you change the header in the Machine Settings screen. Changing the g-code flavour will not change the actual amount of material used so it should be good for measuring the filament usage for your project.

Good luck! I loved those bridge building competitions at school. I won two of them.

Tedsond, 2020-10-02:
Thanks! I can now try and calculate the weight more accurately, this will help me save material in the long run. Thanks for the workaround.

Liger0, 2020-10-06:
Why closing it? It's a report page, not a personal help ticket where it's done when a workoround is given to a single user.

Tedsond, 2020-10-06:
Sorry, I'm not familiar. They said they'd like to add it later. But yea, the workaround causes the printer to fail prints. so I have to switch the Gcode types around all the time. (UM3 doesn't recognise Marlin?) So it's not ideal.

closed 2023-07-13 by github-actions[bot]