Ultimaker/Cura #8449: 4.7.1 Crashes When Adding New Non-Networked Printer

rrileypm, 2020-09-26:
I'm trying to add a Creality Ender 3 to Cura.  This consistently happens every time.  

**Application version**
4.7.1

**Platform**
Windows 10

**Printer**
Anycubic i3 Mega Selected
Trying to install a Creality Ender 3

**Reproduction steps**
1. 3 printers already installed.  Anycubic i3 Mega, Kingroon KP3, Custom FFF printer.  Attempting to install Ender 3.
2. Settings, Printer, Manage Printers, Add Printer, Add Non-Networked Printer.

**Screenshot(s)**
NA

**Actual results**
Cura hard crashes when I click the Add Non-Networked Printer button.  I actually did get to the list of selectable printers once, but when I selected the Creality Ender 3 and clicked the Add button...Cura crashed.
[cura.log](https://github.com/Ultimaker/Cura/files/5286933/cura.log)

**Expected results**
New printer added

**Project file**
No project started yet.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5286933/cura.log)

**Additional information**
(Extra information relevant to the issue.)

rrileypm, 2020-09-26:
I confirmed this problem is isolated to 4.7.1 because reverting back to 4.6.2 solved the problem.

Removing all of the plugins from 4.7.1 did not change the crashing issue.

rrileypm, 2020-09-26:
After adding the new printer in 4.6.2, I tried re-installing 4.7.1 and the new printer did not appear on the list of printers.  I'm switching back to 4.6.2 for the time being until this bug is fixed.

nallath, 2020-09-28:
> Removing all of the plugins from 4.7.1 did not change the crashing issue.

You didn't disable them. The logs show that the startOptimizer plugin is still active. The logs also indicate that a certain file that the ender needs is not available. I suspect that that file is prevented from being loaded by the optimizer. 

So disable the optimizer and try again.

closed 2020-09-28 by nallath

fieldOfView, 2020-09-28:
> So disable the optimizer and try again.

And if that worked, please open an issue at https://github.com/fieldOfView/Cura-StartOptimiser/issues

rrileypm, 2020-09-29:
After disabling the optimizer...then uninstalling the optimizer completely and upgrading to 4.7.1, the problem persisted.

I uninstalled 4.7.1 and went to my  c:\user\ME\appdata\local\cura folder and renamed it cura_old.  I then reinstalled 4.7.1 and it worked fine.  Something must have gotten messed up in cache.

So for me, the problem is resolved.

fieldOfView, 2020-09-30:
Did you rename `c:\user\ME\appdata\local\cura` or `c:\user\ME\appdata\roaming\cura`? Both exist, and the distrinction is important. The former only has cache, whereas the latter has all of the configurations in it.

rrileypm, 2020-09-30:
I renamed it cura_old. The settings were intact when I ran 4.7.1.