Ultimaker/Cura #8451: Mirror Project File Bug

zerspaner-gerd, 2020-09-26:
**Application version**
Cura 4.7.1
(Tested with 4.6.1 and smartavionics Master Version same behavior)

**Platform**
Windows 10

**Printer**
no matter

**Reproduction steps**

1. Loading stl
2. Mirror
3. Save as a project file
4. Load project file

**Screenshot(s)**
[image]
**STL loaded
Everything OK**
[image]
**STL mirrored
Everything OK**
[image]
**After opening the project file
Broken**

**Actual results**
STL is now defective, no longer printable

**Expected results**
Should be loaded exactly as saved

**Additional information**
I have tried it with different STL, all behave the same

nallath, 2020-09-28:
Why would this make the STL not loadable anymore? It just flipped the normals, which is only used for rendering.

As far as I can see this is just a visualization issue. Did you try and print it?

zerspaner-gerd, 2020-09-28:
Since when have Cura upside down normals not bothered?

> As far as I can see this is just a visualization issue. Did you try and print it?

I meant that with me no infill was printed more, unfortunately I have overridden the original project file.
Now I wanted to recreate it and could not reproduce it, so only wrong areas red but is printed correctly

Ghostkeeper, 2020-09-29:
> Why would this make the STL not loadable anymore? It just flipped the normals, which is only used for rendering.

Indeed the normals _should not_ affect the print, but they do, through the winding order of the triangles. It's a known issue for us (CURA-6928).

I'm not able to reproduce this issue though. This is a recording of me executing your reproduction steps: http://dulek.net/work/mirror_reload_project.webm
Maybe it still depends on the model? What model are you using?

zerspaner-gerd, 2020-10-01:
Do you have a MAC?

Maybe just a Windows problem?

Here is a link in it is an STL, two project files and a video
https://www.dropbox.com/sh/tj988x8n0kzdmrs/AAA3qiYIm-NbvyCR0dkhW05Ya?dl=0

But for me it is with every STL!
However, it is currently being sliced correctly, I don't know what led to it that no infill was generated for me.

Ghostkeeper, 2020-10-05:
I don't have a Mac. This was tested on Linux.

Thanks to your video I can now also reproduce the issue. The problem with my previous test was that it didn't display the red overhang areas since the surface was too steep for it to be classified as overhang. And by coincidence the stitching happened to work out well so that the model did get properly filled.

Vandrasc, 2020-10-07:
Devs see internal ticket CURA-7761 for our own reference.

closed 2024-12-08 by github-actions[bot]