Ultimaker/Cura #8453: [4.7.1.AppImage] Segmentation Fault

ljkenny, 2020-09-27:
**Application version**
4.7.1.AppImage

**Platform**
Dell Precision 5500

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.1 LTS

**Printer**
N/A - Cura doesn't even start

**Reproduction steps**
```
~$ ~/Downloads/Ultimaker_Cura-4.7.1.AppImage 
Fontconfig warning: line 5: unknown element "its:rules"
Fontconfig warning: line 6: unknown element "its:translateRule"
Fontconfig warning: line 9: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 76: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 84: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 34: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-fonts-persian.conf", line 35: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 5: unknown element "its:translateRule"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 5: unknown element "its:translateRule"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
Segmentation fault (core dumped)
```

**Screenshot(s)**
N/A

**Actual results**
As above

**Expected results**
It should start.

**Project file**
N/A

**Log file**
As above

**Additional information**
None

nallath, 2020-09-28:
Could you also share the log file (as described in the bug template)

ljkenny, 2020-09-28:
Certainly:

```
2020-09-26 16:44:56,418 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-09-26 16:44:56,427 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-09-26 16:44:56,436 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-09-26 16:44:56,442 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-09-26 16:44:56,449 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-09-26 16:44:56,455 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-09-26 16:44:56,474 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-09-26 16:44:56,481 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-09-26 16:44:56,493 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-09-26 16:44:56,500 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-09-26 16:44:56,508 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-09-26 16:44:56,520 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-09-26 16:44:56,529 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2020-09-26 16:44:56,538 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2020-09-26 16:44:56,547 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2020-09-26 16:44:56,566 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2020-09-26 16:44:56,577 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2020-09-26 16:44:56,584 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-09-26 16:44:56,590 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2020-09-26 16:44:56,599 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2020-09-26 16:44:56,646 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-09-26 16:44:56,672 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2020-09-26 16:44:56,683 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2020-09-26 16:44:56,695 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-09-26 16:44:56,706 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-09-26 16:44:56,740 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-09-26 16:44:56,742 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-09-26 16:44:56,745 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-09-26 16:44:56,747 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-09-26 16:44:56,749 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-09-26 16:44:56,751 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-09-26 16:44:56,753 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-09-26 16:44:56,755 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-09-26 16:44:56,759 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2020-09-26 16:44:56,768 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2020-09-26 16:44:56,780 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2020-09-26 16:44:56,795 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
2020-09-26 16:44:56,808 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ScaleTool
2020-09-26 16:44:56,815 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SelectionTool
2020-09-26 16:44:56,827 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TranslateTool
2020-09-26 16:44:56,835 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TrimeshReader
2020-09-26 16:44:56,847 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPReader
2020-09-26 16:44:56,858 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPWriter
2020-09-26 16:44:56,959 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UM3NetworkPrinting
2020-09-26 16:44:56,988 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin USBPrinting
2020-09-26 16:44:56,995 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UltimakerMachineActions
2020-09-26 16:44:57,004 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UpdateChecker
2020-09-26 16:44:57,005 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-09-26 16:44:57,023 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade21to22
2020-09-26 16:44:57,031 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade22to24
2020-09-26 16:44:57,039 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade25to26
2020-09-26 16:44:57,048 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade26to27
2020-09-26 16:44:57,056 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade27to30
2020-09-26 16:44:57,063 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade30to31
[... cut ...]
2020-09-26 16:45:03,031 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultimae4mzcv/usr/bin/resources/shaders/select_face.shader]...
2020-09-26 16:45:03,035 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultimae4mzcv/usr/bin/resources/shaders/default.shader]...
2020-09-26 16:45:03,039 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultimae4mzcv/usr/bin/resources/shaders/composite.shader]...
```

nallath, 2020-09-28:
Hmm. It seems to stop at the moment that it's trying to load a bunch of shaders. But it would be strange if that caused it, since a few of them are already loaded.

We have had issues with SSL connections in the past. What happens if you start Cura with your internet connection disabled?

ljkenny, 2020-09-28:
Disabling networking appears to have no affect.

nallath, 2020-09-28:
Thanks for checking. I'll discuss it with the rest of the team, maybe one of them has an idea.

Ghostkeeper, 2020-09-29:
Well, after it's done loading those shaders it's going to create the main interface. It crashes there without error message to the user, so that means that Qt had an internal crash somewhere.

Maybe you could try executing this in a command line?

    env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura

nallath, 2020-09-29:
Well, it's not quite _done_ loading those shaders. For me, it continues on for a bit with a few more shaders.

ljkenny, 2020-09-29:
> Maybe you could try executing this in a command line?
> 
> ```
> env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura
> ```

Same result.

rburema, 2020-09-30:
A thing you could try is `force_empty_shader_cache = True`, under the `[view]` heading in `cura.cfg` (normally this is a preference, but since it won't start you have to add it manually).

If that doesn't work: You could try to put `opengl_version_detect = force_legacy` (also under the `[view]` heading in `cura.cfg`). This will of course be a workaround, but it'll try to load the legacy shaders, so if the problem is indeed the shaders, there's another change to get it working (and confirm that it's the shaders). It'll lack certain features that way though (preview will show up as lines and 'lay face flat on buildplate' won't work anymore).

ljkenny, 2020-09-30:
Neither of those options, either together nor on their own worked.

rburema, 2020-09-30:
That;s a shame....

The fontconfig errors remind me of [this](https://github.com/Ultimaker/Cura/issues/7836) issue though, which wasn't a crash, but the error message is similar, and there are a few workarounds already. (Though you may have to scroll a bit...)

JoePater, 2020-11-07:
I also got a segfault, on Fedora 33 running their packaged version (4.7.1). Attached is the standard out and standard error. No idea if it's related. The segfault happens every time I close Cura.
[cura.zip](https://github.com/Ultimaker/Cura/files/5504873/cura.zip)

nallath, 2020-11-09:
> I also got a segfault, on Fedora 33 running their packaged version (4.7.1). Attached is the standard out and standard error. No idea if it's related. The segfault happens every time I close Cura.
> [cura.zip](https://github.com/Ultimaker/Cura/files/5504873/cura.zip)
> It doesn't really affect me as it only happens on exit, but you might want to take a look at it. I'll open a separate issue if you want.

So you didn't use the appimage? We only support the appimage. If it works with the appimage and it doesn't work with the fedora packaged variant, you will have to contact the fedora maintainers.

danfoster, 2021-02-02:
Hi @ljkenny  - I'm not sure if you're still having this problem? I had a very similar issue on 4.8.0. You might want to check out #9220 in case it helps?

ljkenny, 2021-02-03:
To be 100% frank with you, I've moved to Prusa Slicer, as it runs beautifully on Linux.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, and we've introduced a lot of changes since this report. I'll be closing this issue. I hope you understand.

You can download the latest version of Cura with a bunch of improvements and a new QT version here:
[https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Apologies for any inconvenience caused by our delayed response.
Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes