Ultimaker/Cura #8455: Empty spaces on z seam

BerkhanB, 2020-09-27:
Cura version: 4.7.1

OS: W10, GPU RTX2060 mobile

Anet A8

**Reproduction steps**
1. I normally sliced Calibration Cube without changing the settings.

**Screenshot(s)**
Turned to this:
[image]

Previously:
[image]

Why there is a gap there and how can I solve it?

**Project file**
[xyzCalibration_cube.zip](https://github.com/Ultimaker/Cura/files/5288111/xyzCalibration_cube.zip)

**Log file**
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5288113/DxDiag.txt)

fieldOfView, 2020-09-27:
Check the "Coasting" setting.

BerkhanB, 2020-09-27:
Thank you, I realized that I enabled coasting before my last prints but I didn't check the preview for that.

closed 2020-09-27 by BerkhanB