Ultimaker/Cura #8456: Fan PWM curve

Tampa, 2020-09-27:
I rebuilt the cooling solution on my printer. As a result sending it a M106 S2.55 is seen as 100% fan speed due to voltage conversion. In the settings I have to set 0.8% fan speed to get to the level of cooling required not to prematurely cool the filament before it has time to fuse together. It would be nice if in extruder settings we could specify a pwm scale from 100%(0-255) down to 1%(0-2.55) so the fan speed percentages again represent the actual speed. Alternatively a way to override the fan speed with a direct entry of what number should be placed in gcode rather than percentage conversion could work as well.

Ghostkeeper, 2020-09-29:
The [documentation](https://marlinfw.org/docs/gcode/M106.html) specifies that the fan speed must range from 0 (fan off) to 255 (100% duty). Not from 0 to 2.55.
It's weird that your voltage conversion is exactly a factor of 100. Are you sure it's not a firmware issue?

If your printer firmware is not according to that specification, you can use a post-processing script to adjust it. But it would be a bit weird to have the PWM scale as a setting since it's really not allowed to scale. From the top of my head, these are the search and replace parameters you can try with two Search and Replace post-processing scripts:

Search: `\nM106 S(\d)(\d+)\n`
Replace: `\nM106 S\1.\2\n`
Regex: Enabled

Search: `\nM106 S(\d)\n`
Replace: `\nM106 S0.\1\n`
Regex: Enabled

I didn't test this. Just trying to guide you to a solution.

Ghostkeeper, 2020-09-30:
We've discussed this feature request with the team but decided to not implement it, because:
* It seems to be specific to a single home-brew printer.
* It's not according to spec. Not the Marlin spec nor RepRap.
* There is a workaround for your specific case.

closed 2020-09-30 by Ghostkeeper

Tampa, 2020-10-08:
It is not really all that home-brew considering all I did was put a Hero Me kit on my Ender 5 Plus. Even with the 24V to 12V buck converter the cooling is so severe that the PLA has no time to bond together. So this curve was more meant to adjust cooling to the need of the material. I guess I will simply work with the small percentile numbers as best as I can and adjust firmware at some point(still using stock Creality firmware). Please do not remove the float number from the input and output as some other slicers have done, which makes them not usable until I change firmware.

Ghostkeeper, 2020-10-12:
The M106 command is currently set at precision of 1 decimal. We have no intention to change that. We would never change the precision of the input setting either.