Ultimaker/Cura #8457: Slow Travel Feed after Filament Retraction

IsqrAh, 2020-09-27:
Hi,

I started seeing slow travel speeds after a retraction occurs after the initial layer during printing. I traced it back to the G code where it doesn't seem to be inserting the travel feedrate after a retraction meaning the following travel commands are carried out at the feedrate of the filament retraction setting (in my case this is much slower than my travel feedrate).

**Application version**
4.7.1

**Platform**
Windows 10

**Printer**
Tevo Tarantula, Marlin V2 Firmware

**Reproduction steps**
Started noticing issue after updating from 4.6.? to 4.7.1

**Screenshot(s)**

In the preview of the sliced model it would seem that the feedrate is correct (300mm/s) so it looks like the interpreter is just failing to insert a new feedrate after a filament retraction.

[image]

**Actual results**
A snippet of the exported GCode at same layer as above screenshot. I expect to see a new feedrate set on line 126459

[image]

**Expected results**
I think this is how the GCode should look:

[image]

**Project file**

[TT_Right_Z_Top_Bearing.zip](https://github.com/Ultimaker/Cura/files/5288415/TT_Right_Z_Top_Bearing.zip)

**Log File**

[cura.zip](https://github.com/Ultimaker/Cura/files/5288446/cura.zip)

**Additional information**

The problem has been observed with several different models.

Ghostkeeper, 2020-09-29:
Yup, I could reproduce this. The problem was that it was tracking the current speed in mm/s, but after a retraction it stored a speed in mm/min. If the move after was exactly a factor of 60 greater than the retraction speed, it would not change the speed.

It'll be fixed for the next version. Thanks for the report!

closed 2020-09-29 by Ghostkeeper

IsqrAh, 2020-09-29:
Excellent, so until the new version is released I just need to adjust my feedrates so they are not a factor of 60 of each other. Easy work around for now.

Glad I could be of help.