Ultimaker/Cura #8462: [4.7.1] Brim Skipped

CarnageMarkus, 2020-09-29:
**Application version**
4.7.1

**Platform**
Windows 10,
 version 1903,
 build 18362.1082,

**Printer**
Clreality Ender 3 Pro

**Reproduction steps**
Downloaded model:
https://www.thingiverse.com/thing:4122379

1. Opened bunch of parts.
2. Scaled them down to 50%. Rotated.
3. Slice.
4. Guldan11.stl is missing brim. (Just this one)

Reimporting said model did not help

**Screenshot(s)**
On the left. Missing brim.
[image]

Settings per model are not changed.

[image]

[image]

**Expected results**
All should have brim

**Project file**
https://we.tl/t-GYIBe3entI

**Log file**
https://wetransfer.com/downloads/17e4753d575069a0cd972a3f992f0d9520200929092105/65e51c6152f2421489756c6fabdb3e6d20200929092105/605dee

konskarm, 2020-10-02:
I think what you are seeing here is a problem of unoptimized profiles in combination with a weird bottom surface of that part. 

Looking at the part which has no brim, it actually has a very irregular shape on its bottom (anomalies that are smaller than 0.1mm) and the object doesn't really touch the buildplate. Support should be generated underneath it and then this support should end up on the buildplate with a brim. Now, since your Layer Height is 0.1mm and the Support Top Distance is set so high by default (0.2), one layer of support may actually be generated under your model, but it is eventually removed due to the high Support Top Distance and, because it is removed, there is no need for a Brim anymore. 

Try raising the part from the buildplate by 0.2mm (with the preference "Automatically drop models on the buildplate" off) and then reduce the support top distance to 0.05 or something. Then there should be enough space for some support to be generated and also for a brim.

Result: 
[image]

konskarm, 2020-10-02:
By the way, really cool model!

closed 2020-10-02 by konskarm