Ultimaker/Cura #8464: Z-seam alignment randomness problem?

viper0078, 2020-09-29:
Application version
Cura 4.6.2

Platform
Windows 10

I've set the Z-seam alignment to random, but it often prints from the same corner over several layers.
I would like to make sure that adjacent layers do not print from the same corner.
Thank you.

konskarm, 2020-10-01:
Hi @viper0078. This sounds more like a bug than a feature. Can you provide a project file where you see the problem occuring?

viper0078, 2020-10-02:
Hi konskarm.
The figure below shows the GCODE file.
You can see that the start of the perimeter is at the same point in layers 1-3.
Note the current mark immediately after [;TYPE:WALL-OUTER].

Layer 1
[image]

Layer 2
[image]

Layer 3
[image]

In the diagram below, you can see that the travel lines (purple) are grouped together with a starting point every few layers.
[image]

In the diagram below, you can see that the travel lines (purple) are grouped together with a starting point every few layers.
[work.zip](https://github.com/Ultimaker/Cura/files/5315400/work.zip)

konskarm, 2020-10-02:
I put it up for discussion with the team. I'll let you know once I have more info.

konskarm, 2020-10-05:
@viper0078 the path always starts at a random corner of the print as you can see in [this line of code](https://github.com/Ultimaker/CuraEngine/blob/0bc96ef1535d161223971d13b2e373418561ea36/src/pathOrderOptimizer.cpp#L225). It just randomly happens for the starting point to stack up due to the model having only four corners. Since we don't really use a seed for the random generation, this is purely coincidental.

viper0078, 2020-10-05:
Hi konskarm.
The purpose of Z-seam alignment random is to avoid generating Z-seams.
Using the random() function is useless if it doesn't accomplish the goal.

I program in C++ and C# at work, so I can imagine a solution to this problem.
The idea is to remember the print start point, and if the return value of the random function is the same in the next layer, get it again.

Ghostkeeper, 2020-10-05:
It sounds like what you are asking for is a feature to spread around the Z seam evenly? For your purpose, if it alternates the Z seam between two vertices every time you'll still have two visible seams. Or if you have a model with slightly higher resolution than your square and two vertices end up quite nearby each other.

If you can think of a solution to spread around the seam evenly, we could accept that. But keep in mind that we're processing these contours asynchronously (so the previous layer might not know yet where the seams are going to be) and the layers don't always have the same shape, so there is not a 1:1 matching between vertices on adjacent layers.

konskarm, 2020-10-05:
I may have been a little too quick in answering that issue without further explanation. Sorry for that. As @Ghostkeeper mentions, in this case it is paralellism that doesn't allow layers to know the state of previous or following consecutive layers. Therefore a layer cannot make the decision on where to start its path based on other layers.

viper0078, 2020-10-05:
First, I was frustrated that it was closed without agreement.

I do not know what is in the Cura program. So you may be unhappy with my proposal.
I would suggest the following approach.

1. store the position of the print start vertex (XY coordinates).
2. at the start of the next layer of printing, if the coordinates of the randomly selected vertex are the same as the coordinates of the vertex memorized in the previous layer, or within a certain distance (say 5mm), the vertex is re-selected.
If the vertex coordinates are compared for the previous few layers, we should not go back and forth between the same two vertices.
There are some considerations to be made when the model is very small or when the number of vertices is small.

I create software for my job. When I encounter a problem, I'm tempted to give reasons why it can't be done, but I try to think of ways to make it happen.

konskarm, 2020-10-05:
I apologize for that. 

Your approach makes sense, but it requires for the layers to be generated sequentially, so that every layer knows the starting position of its previous layer. Unfortunately that cannot be done with the current architecture of the CuraEngine that generates layers in parallel. The starting point of each layer cannot be memorized and shared with other layers, as e.g. layer 2 may be generated before, after, or during the generation of layer 1. 
Technically speaking, we could just check if the starting coordinates for layer 1 have already been generated and then select the starting point of layer 2, taking into account the 1, and if they have not been generated yet, then randomly select one. This can still lead to an aligned z-seam because e.g. layers 1-4 are being generated in parallel and their random starting point coincide again.

viper0078, 2020-10-05:
So each layer of slices is parallelized by multi-threading. I didn't expect that.
So how about the following idea?

The print start position determined by each layer is stored in a hash table. The keys are the layer numbers and the values are the coordinates of the vertices.

1) When the layer starts printing, select the start vertex of the printout with a random function.

2) Check the hash table to see if there are any records for the previous or next layer.
If there are no records, the vertex selected in 1) is determined as the print start point.
If a record exists, the vertex coordinates are compared. If the vertex coordinates are the same or similar, repeat the re-selection.

3) Once the vertex to start printing is determined, record it in the hash table.

The above will at least ensure that the start points between adjacent layers do not overlap.

In the comparison of 2), if the hash table has information on the two previous and two backward layers, you may want to add that as a comparison target as well.

When registering to the hash table, you can use semaphores and other exclusions as appropriate.

konskarm, 2020-10-06:
I appreciate that you taking the time to think of a solution to this!

My comment above was pointing at a similar solution. Unfortunately, consecutive layers that are being processed in parallel will all select random starting points, as they will not find a starting point for the layers around them. Effectively, this will lead to the same initial problem, as coincidentaly they may still end up selecting the same starting point. 

The only way around that would be to take into account how many threads will be working on that and divide the layers into chunks so that e.g. layers 1-8 will be taken care of by thread 1, 9-16 by thread 2 and so on. Generally though, managing the threads at this level adds a lot to the development time and it is not recommended as it will not take full advantage of the CPU's capabilities (since layer 1-8 may finish really fast, but layers 2-16 may take a long time to finish, thus leaving thread 1 unnecessarily idle).

Also, this issue is primarily visible in very tiny models with only a handfull of vertices in each layer. The problem can be very profound for a triangle model, but most of the times it will not be noticeable for models with many vertices in each layer, at which point the random function has way more vertices to chose from.

Ghostkeeper, 2020-10-06:
If we use a thread-safe set, that sort of thing can be arranged with threading. Viper suggested that there'd be a global hash map and all threads add their used seam coordinates to it, and test if their desired seam coordinate is already used. Some data structures allow such operations in a thread-safe manner.

However if you add that the wall may not be perfectly vertical you'll need to do soft checking (e.g. forbid points within a radius), as described by Viper's idea "if the vertex coordinates are the same *or similar*". Hash sets don't work that way. They can only check if it's the same. It's possible by iterating over all vertices or using our HashBucketGrid, but that's not atomic.

And another issue arises with printing a simple cube of more than 4 layers. If all 4 seam positions have been taken, which seam position should you use next? It won't solve Viper's original issue.

viper0078, 2020-10-06:
I'm not very good at English and I use translation software. I apologize if some parts are difficult to understand.

I don't want everything to be perfect.
The hash table will keep track of the vertex coordinates of all layers, but it will only compare to the layer being processed and the adjacent previous and next layers.

The key in the hash table is the layer number, not the vertex coordinates.
For example, when you start printing the fourth layer, it's not a problem because it searches for Key=3,Key=5.

I'd like to see an improvement in the situation of starting at the same vertex over about 10 layers, as shown in the rectangular image. I think we can use hash tables and semaphores to at least avoid using coordinates that are approximated by adjacent layers as the starting point.
If the Z-seam appears on top of that, I'm convinced that it's inevitable, and I'll change my approach, adjusting different parameters, etc.
However, if the random function starts printing from the same vertex across many layers and the resulting Z-seam is noticeable, I can only conclude that the Z-seam random setting itself is unusable.

I print and incorporate into my products parts that have a cone at the bottom of the printout and a rectangle at the top. How about the reasoning that if there are more vertices, they won't occur?　Are you ignoring the needs of your users?

viper0078, 2020-10-07:
I'm deleting it to repost it.

viper0078, 2020-10-07:
I'm not very good at English and I use translation software.
I'm trying to go through the steps of Japanese, English, and then Japanese to make sure the meaning is the same, but it's not perfect.
I'm sorry if there are parts that are hard to understand.

I don't want everything to be perfect. I'd like to see an improvement in the situation of starting from the same vertex in successive layers, like a rectangular image with about 10 layers.
I think a hash table or semaphores could be used to at least ensure that they don't start at coordinates that are approximated in adjacent layers.

The hash table stores the vertex coordinates of all layers. However, only the layer being processed and the adjacent layers are compared.
The key to the hash table is the layer number, not the vertex coordinates.
For example, when determining the starting point of the fourth layer, we compare it to the vertex coordinates found in Key=3,Key=5. Therefore, there is no problem with the process.

If I'm doing the above process and I see a Z-seam, I'll be convinced that I have no choice, and I'll change my approach by adjusting other parameters and so on.
However, if the random function starts printing from the same vertex over many layers and the Z-seam becomes noticeable as a result, I can only conclude that the random setting of the Z-seam itself is meaningless.

I print parts with a cone at the bottom and a rectangle at the top and incorporate them into my product. What about the theory that if there are more vertices, it won't occur?　Doesn't that negate the needs of the user?

viper0078, 2020-11-15:
Whether it's an octagon or a decagon, the starting point is grouped and not random.

Are you using the srand function?

https://stackoverflow.com/questions/3067364/generating-random-numbers-in-c
https://stackoverflow.com/questions/9711076/why-does-rand-always-return-the-same-value

Isn't each thread generating the same random number?
I think it's a common mistake in the use of the rand function, which is common for beginners.

Ghostkeeper, 2020-11-17:
We are using `rand` and `srand`, seeding with the layer number. But it seems we're not using `srand` everywhere so it could be multi-threading at work indeed.

It is an even more common mistake to use cryptographically-secure randomizers for contexts that are not security-sensitive, imo. But the least we can do is seed the randomizer. In some cases the randomizer needs to produce the same sequence of numbers at every slice (randomize infill start) but in some cases it needs to be actually random and produce a different result even if the slice is the same (fuzzy skin). So seeding once before threadpool is created is not a solution. We'll need something more complex.

rdurham, 2021-12-21:
Hello, I'm having a similar issue in that I'm not really satisfied with the randomness of the random Z alignment in 4.12.1.

I generated a few models using the Calibration Shapes plugin to demonstrate.

Dimensional Accuracy Test:
[image]

Calibration Cube:
[image]

Regular cube:
[image]

The cylinders are what I would expect with even distribution of the Z seam.

The cubes are typical of the slicing I encounter. Multiple rows without change.

I hope I am able to revive this discussion and the devs are able to reproduce this behavior. Thanks!

rdurham, 2021-12-22:
To clarify, I'd like to see these more broken up:
[image]

BarsMonster, 2022-02-18:
I can confirm that I also see exactly same issue in 4.11.0. 
Surprisingly, I see this on round parts, where there is infinite number of corners.

Rwide, 2022-05-29:
[image]
The issue remains in Cura 5.0 and it's not good for printing wheels, bearingparts and gears. I would suggest implementing painting/blocking seam functionality like in prusaslicer. That would solve the problem..

BarsMonster, 2023-01-10:
@Rwide seam painting is a huge amount of work. I also worry about gears / bearingparts - where it is hard to paint random seam... Truly random seam is the solution.

pncookson, 2023-01-10:
This seems to be fixed in the latest version.

Regards,

Paul Cookson

BogusF, 2025-01-04:
If this issue is in fact really unfixable, the feature should be renamed into "spread seam", since random, right now, is very misleading and anyone trying to achieve even seam distribution will unknowingly run into this issue and either be fooled, or frustrated, or both, all of which can't be desirable.

My proposition for a solution is to create a helical ghostpath along the z-axis for seam positioning and whenever the thread has to decide, where to put the seam, you move the helix along the z-axis randomly to achieve actual random seam positioning. 
Another solution would of course be to enforce single threaded slicing. I for one would gladly take the time penalty involved if that's what it takes.

BarsMonster, 2025-01-27:
I've just tested in Cura 5.9.0 and issue is still present. 
On square parts - it is not random at all. On round parts - it is mostly concentrated on one side, and still sometimes exactly match on several layers which makes seam visible.

[image]

[image]

viper0078, 2025-01-29:
I got a notification email, so I came to check after a long time, but it still hasn't been resolved, huh?  
I think there are plenty of possible solutions.  
It seems like you could simply determine the start points for all layers first, then split into threads and slice them individually.  
I don't think that kind of time loss would have much impact on the overall processing time.  
Well, I’m not using it anymore, so this is just me venting.

smartavionics, 2025-01-29:
Works for flat sided objects too...

[image]

willstoeckel, 2025-03-06:
also having this issue. all z-seams concentrated to one side of print

[image]

BarsMonster, 2025-07-06:
I've tested it in 5.10.1 and randomness issue is still present:

[image]

GregValiant, 2026-09-10:
Paint on seams are now possible.  I'll close this.

closed 2026-09-10 by GregValiant