Ultimaker/Cura #8466: Semi-transparent bed (Preview tab) rendered as window transparency on Wayland

myfreeweb, 2020-09-29:
**Application version**
4.7.1

**Platform**
FreeBSD, Wayfire, Qt 5.15.0

**Reproduction steps**
1. Go to the Preview tab when launched on Wayland backend

**Screenshot(s)**
[image]

**Actual results**
The semi-transparent bed is rendered as window transparency, showing windows behind the Cura window, like in
https://stackoverflow.com/questions/35245809/qt-opengl-translucent-objects-show-content-from-behind-window

(Wayland uses transparent surface formats for windows by default. You should use blend modes that don't change the framebuffer alpha. Or request a non-transparent surface format but Qt's client-side decorations have transparent corners..)

Ghostkeeper, 2020-10-02:
Wow, weird issue! Stupid standards.

Ghostkeeper, 2020-10-05:
We've discussed this with the team but decided it's not getting priority from us to fix this yet. It's not an issue that really hurts usability, and the proportion of users using Wayland is small. There are more important issues to fix that would benefit more people.

We'll leave this open as a known issue though. Maybe someone comes up with a fix at some point.

valpackett, 2023-06-10:
Don't think it was actually fixed, though I don't currently have the setup to recheck this

MariMakes, 2023-06-13:
Hey @valpackett,

Thanks for getting back to us.
We also don't have access to a setup to recheck if this issue is still present.
That would also make it hard to test if any solution if we could find one.

I'll be closing this issue,  so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I hope you understand.

closed 2023-06-13 by MariMakes

jamesots, 2023-07-04:
I can confirm that this is still an issue in 5.3.1.

I may take a look at the code if I can get it building.

zombielinux, 2023-10-15:
Can confirm the bug still exists.

luka177, 2024-04-14:
The bug exists and is makinf it cura hard to use

luka177, 2024-04-14:
[image]

flaviut, 2024-05-10:
Some context from a similar application: https://github.com/swaywm/sway/issues/6549#issuecomment-926973760

Cura is extremely complicated to build so I can't test it myself, but could this be as simple as adding a call to [setAlphaBufferSize](https://doc.qt.io/qt-5/qsurfaceformat.html#setAlphaBufferSize) in [`OpenGLContext`](https://github.com/Ultimaker/Uranium/blob/b52767d55880a16fba2707fd06e06653a7848759/UM/View/GL/OpenGLContext.py#L23)?