Ultimaker/Cura #8470: Incorrect retraction distance

kingfranz, 2020-09-30:
**Application version**
4.7.1

**Platform**
Ubuntu 18.04, AMD Ryzen 7 1700, 16GB RAM

**Printer**
Creality CR10S Pro V2 with Micro-Swiss Direct drive extruder

**Reproduction steps**
1. Set retraction distance to 0.5mm
2. Noticed large (and fast) movement of extruder during printing
3. Made graph of extruder commands from gcode which clearly shows movements of 6mm
4. Sliced model in Slic3r, and graph shows a max of 1.6mm

**Screenshot(s)**
[image]
[image]

**Actual results**
Incorrect extrution amount and step loss

**Expected results**
See above

**Project file**
[toltst.3mf.zip](https://github.com/Ultimaker/Cura/files/5304739/toltst.3mf.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

kingfranz, 2020-10-01:
Never mind, I found the problem

closed 2020-10-01 by kingfranz

Ghostkeeper, 2020-10-02:
For posterity it would maybe be helpful if you told what the problem was, and your solution to it. Someone may be having the same problem in the future.