Ultimaker/Cura #8471: SOVOL SV02 printer problem

Fabdu57, 2020-09-30:
Hello,

I have two printers at home : SOVOL SV02 and CREALITY Ender 3 V2.
I use on both exactly same profile : [Creality V2  Figurine.zip](https://github.com/Ultimaker/Cura/files/5305831/Creality.V2.Figurine.zip)

Strangely, when i prepare my item with SV02 profile, i have a lot of defect :
[image]

If i make the same item with Ender 3 printer and profile, no problem.
In attachment both 3.m files : 
[Item.zip](https://github.com/Ultimaker/Cura/files/5305869/Item.zip)

I'm not sure, but i think that problem is Cura with SV02 printer.

Can you help me please ?

nallath, 2020-09-30:
The Sovol 2 has coasting enabled by default. The Creality 3 does not.

closed 2020-09-30 by nallath

Fabdu57, 2020-09-30:
So how can i do in this case ?

nallath, 2020-09-30:
You click the checkmark in the setting list next to "Enable coasting" 

[image]

Fabdu57, 2020-09-30:
Like if i enable it, i have still the problem

nallath, 2020-09-30:
You can also try to change the maximum resolution / maximum deviation setting.

Fabdu57, 2020-09-30:
I am sorry but i don't understand.
You close my topic and propose me some ideas.

As i say, on printer creality i don't have this problem.
Is the slicer is the same for ender 3 pro and SV02 in Cura ?

Fabdu57, 2020-09-30:
Finaly i find the solution...
Your firs proposal was good : activation of coasting, then modify "coasting volume" and put 0 instead of 0.06mm². Then my problem dissapear.

fieldOfView, 2020-09-30:
> Like if i enable it, i have still the problem

What @nallath meant was that by default the Sovol 2 is configured with Coasting enabled. If you turn it off in the sidebar (ie: don't HIDE the setting, but uncheck the checkbox in the sidebar), the defects will probably disappear.

fieldOfView, 2020-09-30:
>  modify "coasting volume" and put 0 

That is effectively the same as unchecking the "Enable coasting" checkbox.

Fabdu57, 2020-09-30:
I don't understand.
After make "0" no holes.
If i put again 0.064mm², no hole again.

For me; slicer for SV02 have a problem...

Fabdu57, 2020-09-30:
Problem is solve.
Thanks to nallath and fieldOfView for your help