Ultimaker/Cura #8472: [4.7.1] Missing GUI elements in Cura under linux

gfadini, 2020-10-01:
**Application version**
4.7.1

**Platform**
5.4.64-1-MANJARO 

**Printer**
none

**Reproduction steps**
After installing either the package cura from community or building the cura-git from AUR, I launch the application and I cannot click on proceed in the wizard, the click buttons are missing.
The AppImage works fine

**Screenshot(s)**
[image]

jellespijker, 2020-10-01:
This probably caused cause because you have a newer Qt5 version then Cura supports. That is because the AUR builds it from source. I personally use Manjaro and the AUR package and I came across the same issue.  Unfortunately we only support the appimage.

Please raise this issue with the maintainer of the AUR repository https://aur.archlinux.org/packages/cura-git/

closed 2020-10-01 by jellespijker

gfadini, 2020-10-01:
@jellespijker  then the Qt5 version may be the problem, currently I'm using 5.15

jellespijker, 2020-10-01:
Mind you with a lot of work and patience you can get it to work from source and qt5.18 but I cannot recommend that if you're are just using Cura for slicing.
Only for development purposes. It took me days to get it right.