Ultimaker/Cura #8474: On open, Save and Exports Cura close

alexwbaule, 2020-10-01:
**Application version**
4.7.1-1

**Platform**
Manjaro (Archlinux based)

**Printer**
Ender 5 Plus

**Reproduction steps**
To get error on open, try to open a file using menu (Ctrl + O) (if you drag + drop, works fine)
To get a error on Save, drag and drop some stl file model, Slice it and click in "Save to File".
To get a error on Export, drag and drop some stl file model, Slice it and click in "Export".

**Screenshot(s)**
App Closes

**Actual results**
App Closes

**Expected results**
Open Dialog to Open, Save or Export file.

**Project file**
Any file that you have it.

**Log file**
```
/usr/lib/python3.8/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.8/site-packages/UM/Settings/InstanceContainer.py:53: DeprecationWarning: invalid escape sequence \d
  version_regex = re.compile("\nversion ?= ?(\d+)")
/usr/lib/python3.8/site-packages/UM/Settings/InstanceContainer.py:55: DeprecationWarning: invalid escape sequence \w
  type_regex = re.compile("\ntype ?= ?(\w+)")
/usr/lib/python3.8/site-packages/UM/VersionUpgradeManager.py:98: DeprecationWarning: invalid escape sequence \.
  ".*\.lock",       # Don't upgrade the configuration file lock. It's not persistent.
/usr/lib/python3.8/site-packages/cura/UI/ObjectsModel.py:72: DeprecationWarning: invalid escape sequence \(
  self._naming_regex = re.compile("^(.+)\(([0-9]+)\)$")
[MainThread] UM.Application.initialize [144]: Initializing Ultimaker Cura
[MainThread] UM.Application.initialize [145]: App Version 4.7.1
[MainThread] UM.Application.initialize [146]: Api Version 7.3.0
[MainThread] UM.Application.initialize [147]: Build type None
[MainThread] UM.Resources.__initializeStoragePaths [446]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [456]: Config storage path is /home/alex/.config/cura/4.7
[MainThread] UM.Resources.__initializeStoragePaths [464]: Data storage path is /home/alex/.local/share/cura/4.7
[MainThread] UM.Resources.__initializeStoragePaths [476]: Cache storage path is /home/alex/.cache/cura/4.7
[MainThread] UM.Resources._copyLatestDirsIfPresent [496]: Found config: ['/home/alex/.config/cura'] and data: ['/home/alex/.local/share/cura']
[MainThread] UM.Resources._copyLatestDirsIfPresent [500]: Latest config path: None and latest data path: None
[MainThread] UM.Qt.QtApplication.initialize [144]: Preferences file not found, ignore and use default language 'en_US'
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [118]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [126]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile
[MainThread] UM.Qt.QtApplication.initialize [173]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [180]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [184]: Initializing version upgrade manager ...
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [188]: User package management file /home/alex/.local/share/cura/4.7/packages.json doesn't exist, do nothing
[MainThread] UM.PackageManager._saveManagementData [256]: Package management file /home/alex/.local/share/cura/4.7/packages.json was saved
2020-10-01 16:08:19,673 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin ConsoleLogger
2020-10-01 16:08:19,676 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin FileLogger
2020-10-01 16:08:19,677 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin OBJWriter
2020-10-01 16:08:19,678 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin STLWriter
2020-10-01 16:08:19,684 - WARNING - [MainThread] STLReader.STLReader.<module> [30]: Not loading numpy-stl due to linux issue
2020-10-01 16:08:19,685 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin STLReader
/usr/lib/python3.8/site-packages/cura/UI/CuraSplashScreen.py:73: DeprecationWarning: an integer is required (got type float).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  painter.drawText(60, 70 + self._version_y_offset, 330 * self._scale, 230 * self._scale, Qt.AlignLeft | Qt.AlignTop, version[0])
2020-10-01 16:08:19,708 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin OBJReader
[... 999 lines cut ...]

Thread 0x00007fedb37fe640 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007fedb3fff640 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007fee3532e740 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 844 in run
  File "/usr/bin/cura", line 234 in <module>
zsh: segmentation fault (core dumped)  cura --debug
```
**Additional information**
I started the cura with --debug flag to get more information.

alexwbaule, 2020-10-01:
Reading the logs, searching for "ERROR" i found 2 pip packages that is missing, installed and still the same error.
Packages are:
 - sentry_sdk
 - trimesh

Ghostkeeper, 2020-10-02:
Seems you are using a custom build of Cura, going by the missing packages and all those QML warnings. In particular, it appears as if you're using the wrong Qt version (not 5.10.2).
Could you try to see if the same error occurs when using the AppImage build from Ultimaker?

alexwbaule, 2020-10-03:
Using the AppImage, works well.

Its the "Official Repository App" from Manjaro (ArchLinux)
https://www.archlinux.org/packages/community/any/cura/

Maybe the packeger has build with some wrong libs. 

The QT5 5.15.x version.

Maybe i will open the bug in Archlinux Packages Repo.

nallath, 2020-10-05:
Since it doesn't occur with the appimage, I'm closing this issue. 

If you want this fixed in the build provided by archlinux I suggest you file an issue report there.

closed 2020-10-05 by nallath