Ultimaker/Cura #8475: Cura 4.7 crashes at start in windows 10

Obiwalex, 2020-10-01:
I used to slice with Cura 4.2 version and decided to install 4.7 but it always crashes at start: it shows a blank window...the Cura icon appears in the taskbar and... after a few seconds the window closes and the icon disappear.
I can't use CURA! 
What is happening with Cura? I'm getting crazy!
If someone can help me to fix this issue I would really appreciate!
Thanks in advance for all your helps. 

I have an ASUS  notebook with:

Windows 10 (Home Ed.) 64b
graphic card : Nvidia GeForce GTX 1050Ti
Core processor: Intel (R)Core (TM) i7-8750H CPU @ 2.2GHz
[cura.log](https://github.com/Ultimaker/Cura/files/5314862/cura.log)

konskarm, 2020-10-02:
Logs do not indicate a crash anywhere. The only problem I see is that Cura cannot connect to the Ultimaker API at all (`WARNING - [Thread-1] UM.Logger.logException [110]: urllib.error.URLError: <urlopen error [Errno 11001] getaddrinfo failed>`), which indicates a proxy error. But it still doesn't lead to a crash since this is a handled exception. Other than that, Cura reports that it loads properly. 
Maybe something has gone wrong while updating the old configurations. Could you try from a fresh install? Go to `%appdata%/cura` folder and empty the 4.7 folder (without deleting the folder itself). Then Cura will start from a fresh install without importing your old configurations.

Ghostkeeper, 2020-10-05:
Do you happen to have multiple monitors and that Intel core assigned to render one of them?
This may be a workaround for you: https://github.com/Ultimaker/Cura/issues/7183#issuecomment-592513731

Tattyt, 2020-10-11:
Mine is doing the same thing how do I add to this existing issue without starting  a new thread. thanks tattyt

Tattyt, 2020-10-11:
[DxDiag cura 4.7.1 issue.txt](https://github.com/Ultimaker/Cura/files/5362043/DxDiag.cura.4.7.1.issue.txt)

It doesnt support the Cura.Log.1 file format

Obiwalex, 2020-10-11:
I've fixed the issue. Well...not me but a friend of mine (he's an expert with PC) found that in another forum someone had the same issue with CURA 4.4 and fixed adding the **_opengl32.dll_** (attached at the bottom) file in the installing directory. So I did it and...do not ask me why but now CURA starts ! Wow!
Try to do so and if you fix too please leave a comment.
[opengl32.zip](https://github.com/Ultimaker/Cura/files/5362293/opengl32.zip)

deanporter, 2020-10-14:
Hi Obiwalex, I had the same issue and put the opengl32.dll file in my C:\Program Files\Ultimaker Cura 4.7\ folder and it's now working! Thank you and tell your friend thank you!

argo86, 2020-10-17:
For me it was solved using Obiwalex  method.
Windows 10 build 2004 newly installed, Cura 4.7.1
Thank you!

agenttatdaddy, 2020-11-03:
Using Obiwalex's method did not work for me. It has made Cura unusable and had to uninstall it. The issue now occurs with all versions of Cura that I installed. Even after a clean install of Windows did not fix the issue. I tried the workaround which also did not work. I tried the other methods of altering text files, and still not working. Sad :(. I really liked the Cura. Time to sell the Ultimaker and find a better compatible product and software. My downtime has been over 2 weeks and I cannot afford that.

nallath, 2020-11-04:
> Using Obiwalex's method did not work for me. It has made Cura unusable and had to uninstall it. The issue now occurs with all versions of Cura that I installed. Even after a clean install of Windows [...]

If you don't provide more information than what is already there, there is very little we can do. If the fix that has worked for others doesn't work for you, it's probably another issue.

agenttatdaddy, 2020-11-04:
OK, so this is what I have discovered after pulling most of my hair out. This is not a Cura issue, but a Microsoft issue. This problem only started for me about a month or two ago. I noticed during that time I received a preview update from MS (KB4580364). So, I tested this out. I reinstalled windows, updated everything but that update above, and everything worked perfectly on a dual monitor setup. No hangs, no errors, no crashing even when moving from window to window, closing, reopening, etc. I could not get the error to reproduce. Then I let windows install the update and low and behold the error came back. As soon as I moved it to another window and closed it, it would not open and the same hang and crash happen. I uninstalled the update and everything worked accordingly. I am not savvy enough to figure out what it is in the update that is causing the hangs and crashes but maybe someone else can figure it out. For now, I have left the preview update off my pc and cura has been working fine ever since.

jaimemoramora, 2021-02-13:
Obiwalex method worked. Someone needs to add that .dll to the installer.

Ghostkeeper, 2021-02-19:
opengl32.dll should be part of your Windows installation. I'm not sure if we can include it with Cura because it is different with various versions of Windows (even within patches of Windows 10) and I don't know if it is binary-compatible. I'm also not sure what the legal implications would be of offering up Windows components for download.

Don't know how it could ever be missing from your Windows installation either though.

jaimemoramora, 2021-02-19:
> opengl32.dll should be part of your Windows installation. I'm not sure if we can include it with Cura because it is different with various versions of Windows (even within patches of Windows 10) and [...]

Thanks for clarifying that opengl32.dll is part of Windows. Certainly now knowing that fact it makes sense to leave it out of the installer. As far as why some Windows installs are missing that library is another subject on its own.

sandyjoe, 2021-04-29:
April 2021, Obiwalex fix still works. I had the same issue with Cura 4.9 crashing at initial launch. adding opengl32.dll to Cura 4.9 directory fixed the issue. thanks

Kaioliver, 2021-10-19:
October 2021 The File from Obiwalex Is working yes, but please be Carefull!!! it have 18MB thats not so normal for a .dll File.
After i put it in the Install Directory from Cura its starting but its really slow you cant work with this ! and i created a file to print but the Print stops after a few minutes i checked my usb and try another the same on the USB they are crazy Files that you cant open.
I think maybe i need to slice it again but after a restart from Cura this is not Possible and Finally i got on both USB Sticks the message that i need to delete the usb sticks, all your data will lost.

I Just can say be Carefull and dont use this File !!!!! Thats my experience.
The Problem with the Cura starting i solved with a Simple Update from Microsoft.

Take Care and stay Printing

jelalanne, 2022-03-25:
You guys realize you are downloading a dll file from a perfect stranger, not knowing what is inside, and putting in on your drive with execution rights?

closed 2024-12-08 by github-actions[bot]