Ultimaker/Cura #8477: [4.6.x - 4.7.x] Top layers of tree support are generated in the air

Liger0, 2020-10-02:
**Application version**
4.6 and 4.7 branches, since when the interface layers were forced on even though you manually disable them

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Supports ( tree supports make it more visible)
2. Interface layers disabled

**Screenshot(s)**
[image]

**Actual results**
The top layers, similar to interface ones, are generated in the air. 
This, on my printer which is perfectly calibrated flow-wise, will make the print fail, not attaching to the near tip of three supports

**Expected results**
No "interface-alike" top layers generated in the air

**Project file**
Tell me who to send it to privately, please.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-10-05:
You could send a project file to r. dulek at ultimaker. com if you'd not like to publish a project file publicly. I'll check.

I have a suspicion this is the same underlying issue as https://github.com/Ultimaker/Cura/issues/8218. We've fixed that issue for 4.8 but I see that we forgot to close the Github issue, so I'll do that now.

Liger0, 2020-10-05:
Sent!

Liger0, 2020-10-05:
Tried on master from 2 october posted on the thread, the issue is still somewhat present
[image]

Project sent again.

Ghostkeeper, 2020-10-06:
Hmm, I'm not getting the same result as you. Strange, but maybe this is caused by recent changes to the simplify algorithm (in which case your latest result would probably still occur with different input). This is what I see with the project file you sent me to reproduce it on master:
[image]

Your problem was on the left-hand side. I'm only seeing interface on the right-hand side. You can see better what area gets classified as support interface if you set the interface pattern to concentric with a very small line width:
[image]
This gets rid of whatever anomalies the infill pattern brings along with it (such as lines not fitting in the shape, or the shape fitting right in between infill lines).

At least the original case doesn't happen any more though: 
[image]

Liger0, 2020-10-06:
I mean, in the last photo, on the right side tree branch, the dots on the left don't seem fully safe to print either.

Ghostkeeper, 2020-10-08:
Yes, by the Tree Support Branch Angle setting, the branch is allowed to shift a certain distance with each layer. In the case of those 3 dots, the dots are partially off the side of the branch on the previous layer, and the limited Maximum Resolution also turned the circle below into a hexagon which made the matter worse. But in practice we're really talking about 0.2mm long lines being 0.2mm off (width of 2 human hairs) from the layer below. The tree branch is probably going to wobble by twice that distance anyway. Most likely the blob will be wiped off on the tree anyway and still support the model. I don't think that is a problem. And if it is, you can reduce the tree support branch angle.

Liger0, 2020-10-09:
I still think that there should be a different setting for the support interface than the rest of the support, because it can even have a different material, and different angle may be needed. 
I think that using a lower angle for it to rest correctly on the tree branches would be better.

Plus the issue of the interface that we state previously, which is a very deal breaker for the tree supports usage.

Ghostkeeper, 2020-10-09:
I'm still unable to reproduce the original issue. On the daily build of that time it indeed generates a spot of support interface but that doesn't hang in the air; it only hangs in the air because of the sparse support interface pattern you're using, which is expected behaviour. And the entire spot is gone in our current master, probably because of changes to the simplification function.

There is something to be said for changing the angle with support interface, but that would introduce a circular dependency. Changing the angle changes the positioning of the support branches, and the positioning of support branches changes where support interface is placed, which would then change the angle again. It's also going to be a miniscule effect.

Liger0, 2020-10-09:
From your earlier image, the layer where that interface layer doesn't seem to be linked to a branch support correctly.

https://user-images.githubusercontent.com/2448634/95197468-f256bb00-07d9-11eb-93f4-e2e5c431418e.png

Ghostkeeper, 2020-10-12:
The interface rests on the model there.

Liger0, 2020-10-12:
> 
> 
> The interface rests on the model there.

Ok, but why? It is set not to start there.

Ghostkeeper, 2020-10-13:
No, in the project you sent me about it still happening on the Master branch, Support Placement is set to "Everywhere".
[image]

hrafnkelle, 2020-10-16:
I can generate tree supports starting in thin air by slicing the Hero Me Gen 5 5015 Single Radial Fan Dual Ducts in the 4.8.nightly build from 2020-10-02. I'm using all default tree settings, just enable tree supports.
[image]
Stl file
https://cdn.thingiverse.com/assets/b8/71/d6/67/94/5015_Single_Radial_Fan_Dual_Ducts.stl

Ghostkeeper, 2020-10-19:
I'm also not seeing that result here, but this bug report is about support interface being suspended in mid-air, not about normal tree support hanging in mid-air. Your screenshot shows normal support, which is likely a different bug with a different source then.

Nonetheless that issue also doesn't reproduce for me:
[image]
My suspicion is that you're not using the default settings after all. Or at least that you're using a printer profile which is different from the global defaults, and not telling us which printer profile that is. A project file would still help, but I suggest you create a different issue for it.

Liger0, 2020-10-19:
That would be a new instance of my solved #8218

Liger0, 2020-10-23:
@Ghostkeeper , in 4.8 beta part of the support interface still rests in the air not being calculated carefully. The project is the same as the one I have just sent for the issue #8218 .
[image]

closed 2024-12-08 by github-actions[bot]