Ultimaker/Cura #8478: [4.7.1] Max Comb distance with no Retract disables retraction

lucabartoli, 2020-10-02:
**Application version**
4.7.1

**Platform**
Windows, but probably all

**Printer**
doesn't matter

**Reproduction steps**
1. Fill in retraction values
2. Leave everything else as per default
3. The printer is not retracting
4. Optionally add more than one model to the plate.
5. Printer is not retracting when travelling between models also

**Actual results**
Default value for Max Comb distance with no retract is set to 0 (unlimited) and combing is enabled everywhere by default.
This affects all retractions, basically disabling it, including when printing more than one model on the same plate. 
There are no clues that retraction could be severally limited (or almost disabled) by that feature. 

**Expected results**
A good default value for max comb distance with no retract or something that can warn the user about retraction (maybe in the tooltip of enable retraction?).
It could make sense to make Max Comb distance with no Retract not affecting travels between models.

Ghostkeeper, 2020-10-05:
I'm unable to reproduce this issue. It's retracting fine for me:
[image]
Project file: [max_comb_distance_no_retract_0.zip](https://github.com/Ultimaker/Cura/files/5327847/max_comb_distance_no_retract_0.zip)

There are many considerations for it to decide whether to retract or not. This is in our documentation: https://github.com/Ultimaker/CuraEngine/blob/master/docs/assets/retraction_combing_hop.svg . This documentation also doesn't include the limits to the retraction count which are part of the retraction routine itself and always happen upon every retraction aside from a few hard-coded retractions.

Max Comb Distance With No Retract should only consider the length combing moves. If a combing move is longer than this setting it must always retract. If it is shorter than this setting it may sometimes still retract, e.g. if it crosses walls.

Do you have a project file that reproduces your actual results? Your reproduction steps don't reproduce the problem for me, so I suppose it depends on the model or specific settings that you didn't mention. The issue template asks for a project file for a reason.

czqu90, 2020-10-16:
**Application version**
4.7.1

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
I might have the same issue. Recently I noticed that my prints got extra stringy and oozy. All I changed was switching PC and installing 4.7.1. instead of 4.2.
I usually work on Ender 3 0.2mm default profile with temperatures and retraction set to my material. My Ender 3 has hardware store glass bed, metal extruder arm and SKR mini v2. 
I did my custom retraction test. 210C/60C, 80mm/s. All prints are on 2/4/6/8mm (bottom to top) retractions, three from the left are on 50mm/s, right one is 25mm/s, maximum retraction count is set to 100 (I usually run 10). But why the difference between two middle ones? First from the left had combing set to within infill, second one - not in skin, two on the right - combing is turned off. Distance between centers of towers is set to 20mm, smaller than default max comb distance of 30mm. It seems that Cura treats empty space between towers as infill where it can dump oozing material. I noticed that extruder doesn't retract between towers on most layers. 
[image]

Also if you compare picture of 50mm/s+within infill test from the top to screen cap from Cura, you can match blobs from test's basis to travels outside the model. There's no or close to no blobs on basis on other two 50mm/s retraction tests, 25mm/s is totally base blob free. 
[image]
[image]

lucabartoli, 2020-10-16:
> I'm unable to reproduce this issue. It's retracting fine for me:
> [image]
> Project file: [...]

Your test is retracting because you're travelling from a volume to a separate one, and yes, this happens only when combing.
Attached a project and the print in which you can clearly see the two lines of no-retract while combing.

The two travels in the shape (see the diagonal and the horizontal ones inside the single object)
[image]

They are clearly visible in the print 
[image]

That said, is it a bug? probably not, everything is fine, you set 0 and you get unlimited comb with no retraction.

Is it a bad behaviour? From my point of view yes, because this setting is basically hidden and can produce unexpected (if you don't know about it) results on retractions.
The results of this settings are also visible on the skin.
Moreover, people is struggling with this setting migrating to 4.7.1 and you can save them hours of debugging time by just alerting in the retraction settings

[CFFFP_Logitech_Keyboard_K120_foot_replacement.zip](https://github.com/Ultimaker/Cura/files/5393500/CFFFP_Logitech_Keyboard_K120_foot_replacement.zip)

Ghostkeeper, 2020-10-19:
Indeed this is expected behaviour. The setting description of Max Comb Distance With No Retract also says that it doesn't cause retractions if set to 0.
The default assumption of Cura is that if it can go through the inside of your model it doesn't need to perform a costly retraction. If it's going through infill it won't be visible anyway, whether you retract or not, since the infill is covered by walls and skin on all sides. If it's going through skin it will be visible anyway, whether you retract or not, since what you actually see is a scar from the nozzle scratching the plastic and re-melting it.
This setting allows you to make it retract anyway, intended to prevent underextrusion after particularly long travels, but of course that can still be disabled by leaving the setting at 0. This is expected behaviour. At 0 it will not do retracted combing moves.

If you want to prevent the scar on your skin, I recommend setting Combing to "Not in Skin". The travel move will go around the skin then, through infill if possible or alongside the walls otherwise. It's a good default for many prints, really.

If you really want to turn every travel move into a retracted travel move, but still perform combing to avoid hitting walls, you can set this Max Comb Distance With No Retract setting to 0.001mm. Travel moves shorter than 5 microns should not occur anyway. I would strongly recommend keeping the setting higher than the distance between adjacent skin lines though.

This setting is considered beyond expert-level. It's not even visible if you select the "expert" visibility preset, only with "all". I don't consider the user experience to be a problem. We can expect the users of this setting to understand this. Users of expert-level or lower will keep this to the profile default, which should be tuned well by default.

closed 2020-10-19 by Ghostkeeper

lucabartoli, 2020-10-20:
I know the setting of Max Comb Distance With No Retract solves the issue, but as you told before this setting is pretty much always hidden.
And the problem is infact this setting is totally hidden, with a default value that have effects on the visible parameters (general retraction).
Alerting in Max Comb Distance WNR is useless, because you usually don't see this setting.

The thing is:
Out of the box, Cura produces objects with visible scars on the skin. If you think this is a desired behaviour I don't know what to say anymore.
Otherwise, I think it could be way better to set this setting to a more conservative value (like 1mm) and let the "more than expert" user to decide what to do with this setting.
I see tons of posts by guys struggling with these scars and missing retractions, do you want to tell them it's an hidden feature, not an issue?

Ghostkeeper, 2020-10-22:
Right, so your issue is with the default value of Max Comb Distance With No Retract for whichever printer you're using. We can change the default value for your printer, if you tell us which printer that is and what a better default value is then. If you don't tell us, we can't change the default value either. We can't test setting optimisation on your printer.

For Ultimaker printers the default of this setting is 0. It was found to have no noteworthy effect on preventing scars on the skin, but it does have a negative effect of creating blobs upon the moment of retracting, taking extra time to retract and wearing down the filament for some types of plastic. Our material engineers did find it useful for the CPE family of materials where it's set to 50mm for Ultimaker's newer printers (S3), apparently because CPE leaks more.

The combing mode has more of an effect on scars than this setting.

lucabartoli, 2020-10-23:
Mine is an Eryone Thinker SE, but you don't have a preset printer profile for it and it's based on the general Custom FFF printer.
I can tell the manufacturer for this particular printer default.
I'm telling you because i've seen this problem in general 3dprinting groups, on other printers too, but I don't know the other printers models.
Isn't maybe better to reverse the approach? 1mm for custom FFF printers (for safety) and 0 for Ultimaker and other not affected printers.

Ghostkeeper, 2020-10-26:
We are not at liberty to change defaults for printers, unless we get word from the community that the defaults need to change for their printer. We could change the default for Custom FFF Printer of course.

However currently there are 15 printers in Cura that have changed this setting to something other than 0 in any of their configurations:
* 9 Deltacomb printers change the default to 5 for PETG materials only.
* 2 Jellybox printers change the default to 50 for all configurations.
* 4 Ultimaker printers change the default to 40 for CPE and CPE+ materials only.

That means that for 298 out of 313 printers, the default is 0. I think we should leave the default for Custom FFF Printer also at 0 then, since most custom-built and unsupported printers will need it to be 0.