Ultimaker/Cura #8483: Error on support generation

LordDarkneo, 2020-10-02:
**Application version**
4.7.1

**Platform**
Windows 10 Pro
Graphic card: GTX 1070

**Printer**
CR10-v2

**Reproduction steps**
1. Adding a mudguard on thinigiverse.
2. Reorient the mudguard to make it vertical
3. Slice (with the generate support option)
4. You will see that supports are not in contact with the part to print... It is stopping at layer 337 (and should normally support an edge at layer arounf 360)
Bonus 5. If you try to make a support blobker on this area (to remove the support generated), it just does not care about it and keeps the support!

**Screenshot(s)**
[image]

**Actual results**
Support is just stopping and useless

**Expected results**
Support supporting the part to print(with the desired interface)

**Project file**
https://www.thingiverse.com/thing:4405565

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
[cura.log](https://github.com/Ultimaker/Cura/files/5318715/cura.log)
 to find the log file to upload, or copy a relevant snippet from it.)

nallath, 2020-10-05:
Could you share a project file? This is likely caused by the values of certain settings.

Ghostkeeper, 2020-10-05:
Reduce the Minimum Support Area setting.

closed 2020-10-05 by Ghostkeeper