Ultimaker/Cura #8484: ultimaker cura 4.7.1

battistipaolo, 2020-10-02:
strange condition:
A) - fulfillment enabled.
B) - support disabled.
action: does not fill.
question: where am I wrong?

regards
paolo battisti

battistipaolo, 2020-10-03:
[250_220_petg.zip](https://github.com/Ultimaker/Cura/files/5321351/250_220_petg.zip)

file .zip (problem)

battistipaolo, 2020-10-04:
Hi, 
section the drawing, positioned on line 30,
 see inside: 
you will also see that the filling is missing
regards
paolo battisti

nallath, 2020-10-05:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

Ghostkeeper, 2020-10-05:
Your model is hollow.
[image]
It should not get infill if it's hollow.

closed 2020-10-05 by Ghostkeeper