Ultimaker/Cura #8485: Bridging line direction

tomsutton, 2020-10-02:
v4.6.7
Windows 10
FLSUN QQ-S

[image]

When bridging, the skin first layer lines go in the wrong direction.

Quite new to this. Is it a bug? Or something I can fix in the settings.

Thanks

Ghostkeeper, 2020-10-06:
There is no setting to determine the direction in which it bridges. That's meant to be automatically determined from the model.

If the experimental bridge settings are enabled, it chooses the bridging direction based on the longest unsupported edge. If they are disabled, it uses the older routine of taking the direction from the two farthest supporting pillars. In your case, the experimental one is probably going wrong because the long edge (parallel to your bridging lines) overhangs due to the slope of the walls there. The old routine would choose correctly then, so you can try disabling Enable Bridge Settings setting in the Experimental category.

It's hard to give more details on this. Do you have a project file that you could share so that we may reproduce the problem and maybe verify a fix or workaround?

ZhuJo, 2020-10-12:
I have also noticed this, especially with Line width set above 0.45 mm (with 0.4 mm nozzle). It simply creates bridges that doesn't make any sense from logical point of view.

First image shows bridging at 0.45 line width. It isn't ideal like PrusaSlicer does (90 deg angle from the walls to create shortest path) but at least, somewhat usable (but not at all with PETG)
[image]

Here is the same bridge with 0.5 mm line width. That is clearly completely out of touch.
[image]

closed 2020-10-20 by no-response[bot]

dgrat, 2022-02-14:
>If the experimental bridge settings are enabled, it chooses the bridging direction based on the longest unsupported edge.
I think this is exactly the opposite of how it should be.

Ghostkeeper, 2022-02-26:
With "edge", I meant the edges of the polygon encircling the area that gets bridged, not the individual skin lines. This was an external contribution to Cura, but I think the idea behind selecting the longest edge is to ensure that at least the that long edge is printed optimally. In your case, it appears like the left and right sides of that model get wrongly classified as bridging in general. Perhaps that is due to the geometry of the mesh though, which is hard to see. You didn't provide any reproduction steps either (which is why this issue was closed originally).

I think the theoretical "optimal" bridging direction would be the direction that minimises the longest bridging line. This requires generating the line pattern in all possible directions though, so an approximation can be made by finding the minimum bounding rectangle (not axis-aligned) and using the direction of the short side of that. The width of the minimum bounding box is different from taking the orthogonal direction to the length of the longest bounding box.

CombatVolcano, 2024-08-06:
Also have this problem
[image]
without bridging settings it is much more reasonable
[image]
this isn't really a weird model and I don't know why bridge settings would do this