Ultimaker/Cura #8486: Create option for Quasicrystal Infill

Shanjaq, 2020-10-03:
**Requested Feature**
Always looking for a stronger infill pattern with low density and consistency through all axes, I'm a big fan of Gyroids.  But there might be a new contender:  Quasicrystals!

**Description/Problem**
Gyroid is a great infill pattern but a bit slow to print and rough on the X/Y axis motors due to the constant sine-wave oscillation required to make the layers.  Quasicrystals could possibly be made with just straight lines!

Here is an example using OpenSCAD to generate them, it's not an infill but a 3D mesh, but the code might be useful in building the infill program:
https://www.instructables.com/3D-Printing-Quasicrystal-Shapes/

Here is an article on Quasicrystals:
https://www.britannica.com/science/quasicrystal
Quote:
"Although when first discovered such structures surprised the scientific community, it now appears that quasicrystals rank among the most common structures in alloys of aluminum with such metals as iron, cobalt, or nickel. While no major commercial applications yet exploit properties of the quasicrystalline state directly, quasicrystals form in compounds noted for their **high strength and light weight**, suggesting potential applications in aerospace and other industries."

**Alternatives Tried**
Gyroid is the current best alternative

**Affected users and/or printers**
Anyone wanting to make strong and light-weight parts on any type printer.

**Additional context**
Here is an interesting Veritasium on the evolution from Penrose Tiling to 3D Quasicrystals:
https://www.youtube.com/watch?v=48sCx-wBs34

nallath, 2020-10-05:
But is their high strength and light weight _more_ than the patterns already there?

Ghostkeeper, 2020-10-06:
The article you quote talks about quasicrystals on a molecular level. This is the same level at which [metals form crystals](https://en.wikipedia.org/wiki/Metallic_crystal#Form_and_structure), so like crystals of several nanometres in diameter. These materials without crystals or in this case quasicrystals can be "strong" in the sense of "tough" then because there is no regular pattern and thus no large fault lines in any direction. They can also be weak in other ways; larger crystals are often harder and sometimes tougher because they are packed more densely with high cohesion.

In 3D printing we're dealing with a millimetre-scale level. Important to strength is the non-uniformity of the bead due to internal stresses, as the material is stretched while it was hot and then solidifies in tension. Also important is the direction in which the majority of the lines are drawn (which is how Gyroid gets its soft-but-tough properties). And the way in which the lines join together (e.g. joining 3 lines in one single spot results in underextrusion due to overlapping lines).

The straight lines you're quoting seems to come from an interesting visualisation that Verisasium uses to show that the tiles of the triangle-and-kite pattern line up in straight lines. The pattern itself can't be drawn with only straight lines, nor any other pattern in that video as far as I saw.

I don't think this pattern would beat our current contenders for the hardest pattern (Cubic Subdivision), toughest pattern (Tri-hexagonal) or most evenly distributed pattern (Cross 3D). And I'd not like to work on it until there is research on how this behaves in a 3D printing scale.