Ultimaker/Cura #8487: Print head display

viper0078, 2020-10-03:
**Application version**
Ver.4.6.2

**Platform**
Win10

**Printer**
custom

Hi.
There seems to be something wrong with the print head display that appears when you move the slider bar in the preview.
When the slider bar is increased, the print head in the current position is lighter in color than the print head in the previous position.
[image]

I think the colors are reversed.
If you move the screen even slightly, the print head in the current position becomes darker in color.
[image]

Also, when the slider bar is reduced, the print head seems to be in the wrong position.
[image]

My preference would be for the print head to be displayed in the current position only. Or I would like to be able to optionally specify whether or not to display one previous position.

Best regards.

Ghostkeeper, 2020-10-06:
Yeah, this issue was known to us already. It's a rendering update bug. If you rotate the camera the rendering updates and it shows the more opaque nozzle in the correct position.

It's been like this since 4.4 or so. I'll see if the team thinks it's worth investigating since I don't think that was ever discussed.

zyddst, 2020-10-11:
hello  could you tell me how to add a printer? l have no printer  ,so i can use cura?

closed 2024-12-08 by github-actions[bot]