Ultimaker/Cura #8488: Linetype and color of the preview screen

viper0078, 2020-10-03:
Hi there.
I would like to see the following items in the preview
Display of Retraction Points
Color indication on the bridge

The image is a view of Simplify3D.
Retraction points are shown as circles.
Bridges are shown in yellow.
[image]

It's important to me to see where the bridge is applied.

Best regards.

Ghostkeeper, 2020-10-06:
This issue contains multiple feature requests. I think your main point is about visibility of the bridge? There are other issues of which the retraction visualisation is a duplicate. I'll edit the title because your title is very ambiguous.

Ghostkeeper, 2020-10-06:
The problem with this is that these lines basically fall in two sets of categories: Some of the bridging lines are bridging outer wall lines. Some of the bridging lines are bridging inner wall lines. Some of the bridging lines are bridging skin lines. So a line could be either bridging or non-bridging, and it could be outer wall, inner wall or skin. These two categories overlap.

fieldOfView, 2020-10-06:
At the risk of becoming cryptic, the bridge lines could be dotted, showing both the line-type color and a bridge-color.

viper0078, 2020-10-07:
Thanks for the response. These are not problems, they are my requests.
The means are whatever you want them to be. As long as you can visually confirm that the bridge is applied. The dotted line is also a good idea. Or how about adding a checkbox to show the bridge in the Color Schema screen?
Please consider it.

katherinehackworth, 2020-10-16:
As a suggestion, perhaps the bridging lines could be a shade darker/lighter than the normal color for that line type. That way they're easily distinguishable and clearly which feature they are.