Ultimaker/Cura #8491: 4.7.1 doesn't use NVIDIA GPU despite being set in NVIDIA control panel.

StarNamer, 2020-10-03:
**Application version**
4.7.1

**Platform**
Windows 10 Build 20226 (WIP), NVIDIA GeForce 730M, Intel HD Graphics

**Printer**
Ender-3

**Reproduction steps**
1. Slice model as normal
2. Open preview

**Screenshot(s)**
 
[image]
[image]
[image]
[image]

**Actual results**
Model displays mostly rendered using single lines. Additionally the preview playback control is absent to the layer drawing sequence cannot be checked. Log file shows that 4.7.1 isn't using the NVIDIA GPU despite being set in NVIDIA control panel.
 
**Expected results**
Normal dully rendered model with playback control using NVIDIA GPU. Note, this works fine in 4.6.2 which also installed. And also set to use the NVIDAI GPU.

**Project file**
[Slicing is fine and the model prints correctly]

**Log file**
Edited log files for 4.6.2 and 4.7.1 (removed OctoPrint not connected messages and truncated to last run).
[cura.edited.log](https://github.com/Ultimaker/Cura/files/5322187/cura.edited.log)
[cura.edited.log](https://github.com/Ultimaker/Cura/files/5322188/cura.edited.log)

**Additional information**
Tried reinstalling NVIDIA driver - no change.
Add curaengine.exe as well as cura.exe to NVIDIA control panel settings - no change.

StarNamer, 2020-10-03:
The Windows 10 settings agree with NVIDIA control panel that 4.7.1 should be using the NVIDA CPU.
[image]

Ghostkeeper, 2020-10-06:
We're not selecting a specific GPU to render Cura. We're simply using the OpenGL context that is provided to us by Qt by the [QOpenGLContext class](https://doc.qt.io/qt-5/qopenglcontext.html). There is no way exposed to us by Qt to select a specific GPU.

As far as I know, Qt uses the OpenGL context that Windows uses to draw the screen that the window is located on. This may be a limitation from Qt which is necessary to ensure the API is the same on all their supported platforms. It may just not be listening to this Windows-specific setting. If you have multiple monitors which are drawn with different GPUs, then moving Cura's window to the other monitor and then restarting Cura may provide a workaround.

In any case, since Qt doesn't let us pick and choose which GPU to use, I don't think there is anything we can realistically do to fix this.

StarNamer, 2020-10-07:
Thanks, I was afraid that was the case. I've already tried moving Cura 4.7.1 to a different screen, but it makes no difference. The big puzzle is why 4.6.2 is OK but 4.7.1 isn't using the designated GPU. I think the next think the try is uninstalling and reinstalling.

StarNamer, 2020-10-08:
For future reference by anyone else, I noticed that the Windows Graphics Settings could not find Cura 4.6 although it was int the NVIDIA control panel settings. So I removed Cura 4.7 from both the Windows Graphics Settings and the NVIDIA control and only added it back to the NVIDIA settings. WIndows now does not find either version of Cura in Graphics Settings, but both work with the NVIDIA GPU.

closed 2020-10-08 by StarNamer

Ghostkeeper, 2020-10-09:
Maybe one was installed with the .exe and the other with the .msi? And one of the installers didn't register the application properly with the Nvidia drivers?