Ultimaker/Cura #8492: Ultimaker Cura 2.70 and 2.71 takes unusually long time to load

zipCoder933, 2020-10-03:
Ultimaker Cura 2.7.0 and 2.7.1 takes an unusually long time to load.  Just recently it has taken so long I wondered if was stuck there forever.

I was wondering if this is a bug on your side or maybe a performance issue.

jellespijker, 2020-10-03:
Please fill in the bug template and provide some logs because this information is to limited for.

FYI I just opened up my personal archive and. I didn't notice any significant change for version for Cura 2.7 ( running on Linux )

jellespijker, 2020-10-04:
There should be a Cura folder in the `C:\Users\<usernamer>\Appdata\Local` in here there is a log file located.
Alternatively you can go to the help menu and click on the item show config folder

btw is there a reason why you are still using 2.7.0 and 2.7.1? We're currently on version 4.7.1 and I estimate that Cura 2.7.0 was released somewhere in 2017

closed 2020-10-17 by no-response[bot]