Ultimaker/Cura #8493: The print parameter description (pop-up display) is interrupted.

viper0078, 2020-10-04:
**Application version**
4.6.2
**Platform**
Win10
**Printer**
Custom

The print parameter pop-up message is interrupted. I would like the message to be displayed without interruption.
Also, is it possible to reference these descriptions as a text file? I would like to convert this using a translation software.
Best regards.

nallath, 2020-10-05:
This is caused by the print setting guide and not by Cura itself. That being said, this was semi recently changed. If it's not in the 4.7.1 release, it will be in the 4.8 release.

closed 2020-10-05 by nallath

Ghostkeeper, 2020-10-06:
Yeah that was merged into 4.8 (while 4.7 was in beta). I purposefully waited until 4.7 was branched off and feature frozen to allow more time to test the scrolling of tooltips.

viper0078, 2020-10-07:
Thanks for the answer.

By the way, is it possible to refer to these detailed instructions in a text file or something like that, I wish I could do it on the web site.
I want to convert them in my translation software.

I'm taking the following steps
1) Pop up the detail view.
2) Capture the screen and display it in paint software.
3) Convert it into a text file with OCR software.
4) Use a translation software to convert the file.

It's very complicated.

Ghostkeeper, 2020-10-09:
The Settings Guide is available online here: https://github.com/Ghostkeeper/SettingsGuide

You can view the articles there online too (under "resources/articles"). Or you can download the text files by clicking on the green "Code" button on the top right corner and then "Download zip".

viper0078, 2020-10-09:
Thank you for the information.