Ultimaker/Cura #8505: Keyboard shortcut for Move tool doesn't work

Dignan17, 2020-10-06:
**Application version**
4.7.0
(The version of the application this issue occurs with.)

**Platform**
Windows 10 fully updated as of 10/6/20. Not sure of GPU but doesn't seem related to this issue. Integrated Intel graphics I'm sure.
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Anycubic i3 Mega
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
Keyboard shortcuts are invaluable to me for saving time and aggravation. I can hit documented keys to select other tools like rotate, but the move shortcut, which I can only find listed as Q, doesn't work.
2. (Something you did next.)
Tried switching to it from various tools. Q does nothing. Tried to research shortcuts more, couldn't find anything. The software also doesn't note keyboard shortcuts.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Nothing

**Expected results**
(What should happen after the above steps have been followed.)
The move tool is selected

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-10-07:
The keyboard shortcut for the move (or translate) tool is T, once your model is selected. I just tried this on Windows 10 and it seems to work fine.

Dignan17, 2020-10-07:
Oops! I searched a ton and never found that shortcut. So sorry!

But thank you!

closed 2020-10-07 by konskarm

Ghostkeeper, 2020-10-09:
The shortcut should normally be shown on hover, but the tooltips of those tool buttons broke in 4.7.