Ultimaker/Cura #8506: Random Changes in Settings

girron, 2020-10-07:
I just updated to 4.7.1 and I have been experiencing random issues with the settings changing. An example would be on a profile that I have, the bottom line count increased to 999 while the top line count changed to 0. This was causing issues with the print and it took me awhile to figure out what was going on. 

Other items in the settings will randomly change as well causing me to have to switch profiles and then switch back and discard the changes that Cura made.

konskarm, 2020-10-07:
This is quite strange. But please, fill in the entire [bug template](https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+Bug&template=bug-report.md&title=) with logs and reproduction steps so that we have enough information to identify the issue.

zyddst, 2020-10-11:
hello could you tell me how to add a printer?if l have no printer ,l can not use cura?
thank you
https://github.com/zyddst/restudy/blob/master/Screenshot%20from%202020-10-11%2020-01-07.png

closed 2020-10-21 by no-response[bot]