Ultimaker/Cura #8507: Rotating Model Making the Scale Changes !!!

Ruweshed, 2020-10-07:
When i try to Rotate my Model to Fit Within the Printing Area Which Should have Computed by some Button and you Guys missing That Feature,,,, Anyway when i Rotate it on X or Y access its Changing the SCALE of the WHOLE MODEL!!!! WHYYYYYYYYYYYYYY 

Guys you dont have an Automatic  Fitting Button and even Wont let US Rotate with our Same %100 Model Size !!!!!!???

This have to GET FIXED !!

konskarm, 2020-10-07:
This sounds like a duplicate of https://github.com/Ultimaker/Cura/issues/4136. Nevertheless, please fill in the entire [bug template](https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+Bug&template=bug-report.md&title=) so that we can identify the problem clearly.

Ruweshed, 2020-10-07:
Well anyway this could be fixed by Adding some Feature to Fit the Model "AS IT IS" within Printing AREA... i mean Within the PRINTABLE ALLOWED AREA ... is there any Project for such Feature soon??

konskarm, 2020-10-08:
We are currently working on changing our arrange algorithm to a new version, which should deal with this exact problem. I'll close this issue, since the original bug (rotation causes scale to behave weirdly) has been reported again and we want the discussions of similar issues to exist in a single thread (#4136).

closed 2020-10-08 by konskarm