Ultimaker/Cura #8512: [7.1.0] Draft Shield ignores build Volume

Waldkaeuzchen, 2020-10-09:
**Application version**
7.1.0

**Platform**
Windows 10

**Printer**
Ultimaker 3

**Reproduction steps**
1. Enable Draft Shield

**Screenshot(s)**
[image]

**Actual results**
The draft shield is generated outside the buildvolume of the printer

**Expected results**
Draft Shield should fit onto the buildplate

**Project file**
[UM3_Main base.zip](https://github.com/Ultimaker/Cura/files/5355313/UM3_Main.base.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5355318/cura.log)

zyddst, 2020-10-11:
could you tell me how to add a printer?
l have no printer,so l can't use cura?

Ghostkeeper, 2020-10-12:
Ah yeah, I see. The problem is with the combination of Tree Support or Conical Support, and Draft Shield / Ooze Shield / Brim / Skirt.

You see before slicing it's not yet known where the tree support will end up on the build plate. It could branch out and land besides the model and then be placed outside of the build plate. So we're [adding](https://github.com/Ultimaker/CuraEngine/blob/0bc96ef1535d161223971d13b2e373418561ea36/src/TreeSupport.cpp#L677) the edge of the build volume to the collision areas that Tree Support needs to avoid. However if you're using a draft shield that still gets put around the support with a certain distance. So then the draft shield still goes out of the build volume.

The same goes for anything else that gets put around support, including brim. Conical support has a similar routine.

Instead of having the original shape of the build volume as collision area, we should maybe use an inset of that which requires a certain distance to allow space for the brim/draft shield/etc. A bit maintenance-heavy maybe though. The calculation for that in the front-end is quite involved.

Vandrasc, 2020-10-13:
Devs see CURA-7768 for our internal reference.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

Waldkaeuzchen, 2024-11-21:
I just sliced the project again using the current version (5.9.0), and the problem still seems to be there.
[image]