Ultimaker/Cura #8513: [Feature request] Add sync settings between others PC

userosos, 2020-10-09:
Hi! I use 2 different laptops and it would be good if you added synchronization of my settings between different laptops. Backing up addon is not convenient.
Add the synchronization settings between different PCs.

Ghostkeeper, 2020-10-12:
How would you envision that this looks from a user's perspective?

When I'm envisioning how a sync of configurations between computers would work, I'd imagine that:
* You need to log into your Ultimaker account to access the configurations.
* It doesn't automatically overwrite any existing profiles on your computer, but is a manual action with a warning that it could be destructive.
* You'd want some way to differentiate between the configurations that were on various computers when the syncing started. Like a list of configurations.

When I write it down like that, it sounds pretty much exactly like the back-up plug-in. Except that the back-up could be placed more prominently maybe. It's currently kinda hidden.

Ghostkeeper, 2020-10-12:
Actually, there was already a discussion going on here, I see: https://github.com/Ultimaker/Cura/issues/8513

alvgalrus, 2021-02-13:
@Ghostkeeper You marked this as duplicate of itself. I signed up for an Ultimaker account but settings won't sync between Ubuntu and Windows installations.

alvgalrus, 2021-02-13:
@Ghostkeeper I believe you marked this as duplicate of itself. I signed up for an Ultimaker account but settings won't sync between Ubuntu and Windows installations.

Ghostkeeper, 2021-02-19:
Oops! Probably a copy-paste mistake, but I can't find the original one which I thought this was a duplicate of either indeed.

Moving configuration between different operating systems is indeed not supported. The problem there is that these different operating systems have different requirements as to where the different types of data get stored. On Windows, we store the cache in `%USER%\AppData\local` and configuration and data in `%USER%\AppData\roaming`. On MacOS, we store everything in `~/Library/Application Support`. On Linux, we store the cache in `~/.cache`, the configuration in `~/.config` and the data in `~/.local`. These distinctions make it difficult to restore a configuration on a different computer.

closed 2021-02-19 by no-response[bot]