Ultimaker/Cura #8514: Modify support structure

ericlmccormick, 2020-10-10:
I have used Cura, Sli3er, PrusaSlicer and UpStudio. Of all of them, I find that UpStudio has superior support but only works on their hardware. Whatever they do creates almost perfect circles where others have an oval. Additionally, the support structure hangs out a little bit from the area it is supporting. Doing this gives an area to grab and remove instead of having to dig into the part.

zyddst, 2020-10-11:
hello could you tell me how to add a printer ? if l have no printer ,l can't use cura?
https://github.com/zyddst/restudy/blob/master/Screenshot%20from%202020-10-11%2020-01-07.png

Ghostkeeper, 2020-10-12:
This feature already exists in Cura. It's called Support Horizontal Expansion: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_offset.md

closed 2020-10-12 by Ghostkeeper