Ultimaker/Cura #8518: Parts of 3mf file grayed out in object list and gray yellow in prepare window

martijnschouten, 2020-10-11:
**Application version**
Cura 4.7.1

**Platform**
Windows 10
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Diabase H-series

**Reproduction steps**
1. generate 3mf file with solidworks with 3 different objects
2. load the 3mf file in cura
3. move the 3mf file in z to a height of 0
4. move the 3mf files to the center of the bed
5. ungroup the 3mf file

**Screenshot(s)**
[image]

**Actual results**
Two parts are grayed out in the object list and gray-yelllow in the prepare window. These parts are not sliced.

**Expected results**
They are not grayed out, have the color of the selected filament and are sliced.

**Project file**
[source files.zip](https://github.com/Ultimaker/Cura/files/5361933/source.files.zip)

**Log file**
[log.zip](https://github.com/Ultimaker/Cura/files/5361941/log.zip)

martijnschouten, 2020-10-11:
Alright let me close my own issue. The problem was just that the parts were on the location of the prime tower. Moving the parts solved the issue.

closed 2020-10-11 by martijnschouten

konskarm, 2020-10-12:
You can also move the prime tower itself by setting Prime Tower X and Y values. Although that is a manual process.