Ultimaker/Cura #8519: Import location of stl in an assembly from an stl

martijnschouten, 2020-10-11:
**Is your feature request related to a problem? Please describe.**
I want to export my assembly from Autodesk Inverter to Cura. This assembly contains 7 parts and i don't want to manually have to allign all stl's. Also some parts need to be printed from a different material. If i import the assembly with the Inventer plugin I can't select the individual parts. Inventor as far as I know can't export in the 3mf format.
[image]
[image]
[parts.zip](https://github.com/Ultimaker/Cura/files/5361994/parts.zip)

**Describe the solution you'd like**
If I export an stl for all the parts in Inventor and open this in Simplify3D they are all alligned as they were in the assembly in inventor. If i do the same in cura they all end up with their center on top of each other I have to allign them manually.

**Describe alternatives you've considered**
- export stl's individually and align the manually -> time consuming, especially since this is only a small test structure and I want to go to a design with even more parts.
- export 3mf file -> not supported by inventor
- Try drawing in solidworks and export an 3mf file -> I prefer drawing in inventor.

nallath, 2020-10-12:
The inventor plugin isn't created by us, so any issues should be discussed there. The official thread can be found here: https://community.ultimaker.com/topic/21489-inventor-plugin-official-thread/

closed 2020-10-12 by nallath

martijnschouten, 2020-10-12:
Check thank you modifying the inverter app indeed would solve the issue. But you as the title of the issue says, I would like to see an option in cura to not manually reassign the origin of my stl's.

fieldOfView, 2020-10-12:
Could you try selecting all imported models in Cura and merging them? You can ungroup them afterwards.

martijnschouten, 2020-10-12:
Thank you fieldOfView, merging all models does align then as how they were aligned in the assembly in the inverter. This solves the issue.