Ultimaker/Cura #8521: Dual Extruder M106 and M107 command malformed when both extruders use same part cooling fan "0"

c3D-Dan, 2020-10-11:
Dual Extruder setup.  M106 and M107 command malformed when both extruders use the same part cooling fan "0"

When gcode is generated, extruder 1 fan gcode appears as follows:
M106 S255 P0

Extruder 2 fan gcode appears as follows :
M106 S255

When Extruder 2 has a unique fan number, gcode is generated correctly.  

Although not ideal, current work around is to search and replace 
"M106" 
"M106 P0"

nallath, 2020-10-12:
But "M106 S255" is a correct g-code. It's even listed as an example in the reprap g-code list. https://www.reprap.org/wiki/G-code#M106:_Fan_On

c3D-Dan, 2020-10-12:
Only if your printer has 1 fan.  If your printer has more than one fan defined, it needs a fan number.

c3D-Dan, 2020-10-12:
Using Marlin 2 bugfix by the way.

nallath, 2020-10-12:
Nope: "Pnnn Fan number (optional, defaults to 0)"

Since it's an optional parameter, not setting it should mean it defaults to 0.

c3D-Dan, 2020-10-12:
This is pretty easy to test and confirm.  Please Install Marlin 2 bugfix and test.  

There's no reason not to include the fan number in the code.  Clearly not doing so makes Cura less compatible with arguably the most popular firmware.

nallath, 2020-10-12:
It's not that I don't believe that the issue exists ;) What I am saying is that the way I see it, this is an issue with Marlin, not with Cura. The older versions of Marlin did accept this, since that is what we've been using since then. So it's a bit weird that Marlin suddenly added such a breaking change! 

I also can't install marlin 2, because I don't have a printer that would work with it. It would take me a lot of time to get it to work on an UM3.

Ghostkeeper, 2020-10-13:
Marlin's [own documentation](https://marlinfw.org/docs/gcode/M106.html) also indicates that if no fan index is given, the print cooling fan is used.
In Marlin's source code you can read that it should take the `_ALT_P` if there is no `P` argument to the `M106` command:

https://github.com/MarlinFirmware/Marlin/blob/8e1ea6a2fa1b90a58b4257eec9fbc2923adda680/Marlin/src/gcode/temp/M106_M107.cpp#L58

And `_ALT_P` is defined as the fan belonging to the active extruder, unless the printer has only one fan for all extruders:

https://github.com/MarlinFirmware/Marlin/blob/92076c65601de5fdfe0facf32dd60c57aab05747/Marlin/src/gcode/temp/M106_M107.cpp#L35-L41

So from all that I can see, the intended behaviour from Marlin is that it takes the fan on the print head if `P` is not supplied, even in the 2.0.x-bugfix branch of Marlin. That would make this a bug in Marlin.

closed 2020-10-13 by Ghostkeeper

c3D-Dan, 2020-10-13:
Just because this isn't specifically a CURA issue, doesn't mean it's only any issue with Marlin.  If no extruder fan number is defined in machine settings, then I would completely understand.  Not also including the "P" argument when Fan 0 is defined reduces the capacity of post processing scripts to manipulate the resulting gcode. I really hope you reopen this.

Ghostkeeper, 2020-10-13:
Cura can accommodate for differences in firmware to a large extent, but building a feature for bugs like this is out of scope. It would involve an inordinate amount of work and add unnecessary complexity for something that only a limited number of people will use for a limited amount of time.

If there is a way to build this where Cura is
* still adhering to the g-code specification,
* not any more complex to use (i.e. no additional options),
* not needlessly adding to the g-code size for the majority of users,

and you think that is simple functionality, you are free to add that to Cura.
To help you build it, I think you should first look to this function here, which writes the fan command: https://github.com/Ultimaker/CuraEngine/blob/0bc96ef1535d161223971d13b2e373418561ea36/src/gcodeExport.cpp#L1143-L1180
I don't really have a good hint for how to build this with good UX though. If you have any ideas, we'll welcome a pull request that shows us.

Ultimaker won't spend time on this to develop it for you though.

Ghostkeeper, 2020-10-13:
I don't see a bug report about this yet on the Marlin firmware repository. So you might be the first to discover this bug. Please report it to them! They will not fix your bug otherwise. And they might come up with a workaround for you, too.

c3D-Dan, 2020-10-13:
More text has been consumed in this discussion that will ever be consumed by adding a "P" argument to the a fan defined as 0. It's not like they turn in and off 500 times a layer.  Seems it would make your gcode clearer to ignore pointless gcode "gains" like this.  Simplification gone mad IMO.  

I can see how it's technically not your problem though.  Probably got better things to do.  Fair enough.