Ultimaker/Cura #8522: Cura 4.7.1 Slicing Preview Play only at Top Level

chiayc, 2020-10-12:
**Application version**
Cura 4.7.1 for Mac

**Platform**
MacOS 10.15.7 (Intel Iris Plus 1 GB)

**Printer**
TwoTrees Bluer Plus

**Reproduction steps**
1. Open .STL file
2. Slice the model
3. Go to Preview
4. Select a vertical layer (not the top in this case), click Play button at the bottom

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
Only show slicing of the top layer

**Expected results**
Expected slicing of the layer specified in the vertical selection bar

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-10-13:
I'm not able to reproduce this problem. When I try this, it just starts at the layer where I left off with the slider. Here is a (sped-up) recording: http://dulek.net/work/playing_halfway_layers_timelapse.webm

It should be noted that it starts playing from the top of the range of visible layers. If you only adjust the bottom handle of the slider, then the top of the range is still the highest layer and so it will start playing the last layer. This is expected behaviour. Is that what you meant? What if you adjust the top handle instead of the lower handle?

chiayc, 2020-10-13:
Hi, thanks for the response. I have adjusted the top handle (the one controlling the vertical layer), even if I select a middle layer, when I click the Play at the bottom handle, it still only play the top layer. You can actually see from the file I attached that I have selected a middle layer from the top handle but it still show the printing preview of the top layer.

Regards,
Chia

chiayc, 2020-10-13:
Sorry, I got it now, my mistake, you are right, I didn't see the top handle, I thought what you mean was vertical sliding ruler. Now I got it to work, thanks.

closed 2020-10-13 by chiayc